biemster/FindMy #2: iCloud keychain access error

xliee, 2022-07-17:
MacOS Version: Big Sur (11.5.2)
I can access the different keychains well with the security command but the Icloud keychain.

biemster, 2022-07-17:
so you can get the `search-party-token`  but not the `account UUID`? Is the machine logged in to icloud, and are you logged in via ssh or direct?

xliee, 2022-07-17:
Im logged in the Mac normally with the account I have in the Mac, then open the terminal and try to run the security command.
In the Keychain Access app I see 2 party tokens:
`com.apple.account.DeviceLocator.search-party-token`
`com.apple.account.AppleAccount.search-party-token`
Not able to access any of those neither any of the iCloud keychain keys, but no problem accessing other keychain keys.
Im not logged in via ssh just the terminal app, with and without root account

biemster, 2022-07-17:
Is the openhaystack app working correctly for you?

xliee, 2022-07-18:
> Is the openhaystack app working correctly for you?

Yes it is. Built the mail plugin and a custom server fot ehr openhaystack and everything works well, but would like to get it working without the mail plugin
If you want you can contact me on telegram or discord to discuss about this
discord: xliee#8727
tg: @xliee

biemster, 2022-07-19:
I'm working for a couple days now already to remove the keychain access commands, and the appleID UUID I can get from `~/Library/Preferences/MobileMeAccounts.plist`. But the `search-party-token` I did not find yet unfortunately, so this does not fix your issue fully yet.

vekexasia, 2022-08-12:
Hey @biemster nice work on this.

I'm having the same issue (I think)

```
security: SecKeychainSearchCopyNext: The specified item could not be found in the keychain.
Traceback (most recent call last):
  File "request_reports.py", line 60, in <module>
    searchPartyToken = getSearchPartyToken()
  File "request_reports.py", line 40, in getSearchPartyToken
    return subprocess.check_output(['/usr/bin/security', 'find-generic-password', '-w', '-s', 'com.apple.account.AppleAccount.search-party-token']).rstrip('\n')
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['/usr/bin/security', 'find-generic-password', '-w', '-s', 'com.apple.account.AppleAccount.search-party-token']' returned non-zero exit status 44

Not sure this has any value but your few lines of code here:
```
#!/usr/bin/python

import objc; from Foundation import NSBundle, NSClassFromString

AOSKitBundle = NSBundle.bundleWithPath_('/System/Library/PrivateFrameworks/AOSKit.framework')
objc.loadBundleFunctions(AOSKitBundle, globals(), [("retrieveOTPHeadersForDSID", '')])
util = NSClassFromString('AOSUtilities')

print(util.retrieveOTPHeadersForDSID_("-2"))

```

provides X-Apple-MD and X-Apple-MD-M

biemster, 2022-08-12:
Hi @vekexasia,
You indeed seem to have the same issue. I'm still working on it, but I don't have a Big Sur box so progress is slow.
I'm not sure what you mean with the second part of your comment, that's the code I posted to the OHS repo so I know what it does?

vekexasia, 2022-08-12:
@biemster i'mtotally unfamiliar with all these apple things. So I thought that maybe for you to know that one thing worked and the other didn't could have given some intel about how to fix it. 

:) Apparently they're unrelated I guess?

biemster, 2022-08-12:
They are not totally unrelated, and I am indeed hoping I can find a `Foundation NSBundle` that can also retrieve the `search-party-token`, but I did not find that yet..

vekexasia, 2022-08-12:
> Hi @vekexasia, You indeed seem to have the same issue. I'm still working on it, but I don't have a Big Sur box so progress is slow. I'm not sure what you mean with the second part of your comment, that's the code I posted to the OHS repo so I know what it does?

anyway @biemster i just installed via proxmox a monterey hackintosh to check out if itwas my installation or not and yeah. It prompts the same issue. If you have a proxmox installation I think you can get it up and running in less than 1h. Or otherwise i can help you iif you need to test stuff... just let me know :)

biemster, 2022-08-13:
> Or otherwise i can help you iif you need to test stuff... just let me know :)

thanks!

biemster, 2022-09-07:
commit message closed it prematurely, @vekexasia and @xliee could you confirm the latest version fixes your issue?

biemster, 2022-09-08:
That's a pity, I was hoping this key was a constant for at least a while. Back to the drawing board.

olivluca, 2022-09-08:
I cannot say, I was using yesterday key and all of a sudden it stopped working.
I think it depends on the time and date (my virtual machine goes out of sync since I just save state in virtualbox).
FWIW, when the new key gave me the 401, I logged out and logged in again and that same key worked (and the time was in sync, with just a 3 minutes offset, both before and after).

biemster, 2022-09-08:
Maybe it is the relogin that changes the key? I'm going to monitor this a bit first before I'm shelving this solution.

olivluca, 2022-09-08:
No, the key is the same, so I'm at a loss as why the same key gave me a 401 before and not after.

biemster, 2022-09-08:
One option would be to add a dependency like https://github.com/n0fate/chainbreaker, then we would only need to enter the keychain password (which is constant). But I'd like to keep the number of dependencies 0, so I call this plan B.

olivluca, 2022-09-08:
I tried it with --dump-all on the keychain you're using to get the iCloud decription key (the one starting wih a digit instead of the one with my username) but it errors out.
Anyway, even if it worked, I think the token is generated on the fly, so I doubt chainbreaker would meet our needs.

olivluca, 2022-09-08:
Oh, right, one is a symlink to the other :roll_eyes:

biemster, 2022-09-08:
The keychain that holds the iCloud decryption key is `~/Library/Keychains/login.keychain-db`, did you try on that? (from your comment it sounds like you are running chainbreaker on `~/Library/Application\ Support/iCloud/Accounts/<dsid>`)

biemster, 2022-09-08:
I was planning on asking for the login password when the script starts, and then read the decryption key using that when needed. Which would have the script holding that password in memory, which is not ideal.

biemster, 2022-09-08:
I'm looking at `https://gist.github.com/williballenthin/d6bf9f1553d9fa27e0cc6880a6d992b4` now to see if I can simplify it (which seems doable), then I just read the login.keychain-db file directly.

olivluca, 2022-09-08:
> The keychain that holds the iCloud decryption key is `~/Library/Keychains/login.keychain-db`, did you try on that? (from your comment it sounds like you are running chainbreaker on `~/Library/Application\ Support/iCloud/Accounts/<dsid>`)

Oh, you're right, now I tried the login.keychain-db, different error but it still errors out.

biemster, 2022-09-09:
As an added benefit, the `request_reports.py` can now be called over SSH! woohoo!

olivluca, 2022-09-09:
It works but it prompts for the password. Maybe now the -k switch should be used for the keychain passphrase?

olivluca, 2022-09-09:
I created pull request #10 for that

biemster, 2022-09-09:
A `ps aux` would show the password then, that's a bit too loose security wise for me.. I think the prompt is fine. People who want to script it should be able to take it from here.

biemster, 2022-09-09:
I will close this issue actually, since the code moved away from the initial issue. If people encounter problems with the new keychain reader, please open a new ticket.

closed 2022-09-09 by biemster

olivluca, 2022-09-09:
I'll just leave a quick note here: I resumed the virtual machine and I got the 401, which fixed itself after a couple of minutes (in the meantime I tried to adjust the clock, which was showing the time when I suspended the machine).
I think that there's a periodic task that's updating the token in the keychain, so maybe that will have some implication in your plan to use the recovery image.
Either that or touching the clock triggered the update.

biemster, 2022-11-22:
The rotating search party token is read from the MMeToken db, @mrx23dot do you by any chance have more than one iCloud account on your machine? what is the content of your `~/Library/Application Support/iCloud/Accounts/` directory?

biemster, 2022-11-28:
It is possible that the script is mixing authentication data between these two accounts to request the reports. So it might use the `search-party-token` from one of your accounts and `anisette` data from the other, resulting in a 401. I don't have multiple icloud accounts to research this problem, how are your python skills and do feel up to diving in the code a bit?

commit 9563c878 (2022-09-07): improve scriptability by abolishing subprocess calls, require only iCloud decryption key as input. should fix #2