biemster/FindMy #6: python2/3 and the httplib

Astu04, 2022-08-02:
Hi! I was rewriting the code to use python3 but I'm getting 401 Unauthorized requests. 
I wanted to ask you if there was an issue with the httplib library, with python3 I have to use http.client instead. 
Or maybe you just wrote the code in python2 due to personal preference?

closed 2022-08-02 by Astu04

biemster, 2022-08-03:
The 401 is almost always the `Authorization` header indeed, well spotted. I wrote the code in python2 because that comes preinstalled on Catalina, and I'm trying to make this work with the Catalina recovery image only so I don't need a full macOS install.

Astu04, 2022-08-03:
I didn't know about the preinstall. I'm trying using a docker debian:bullseye, but I think I'll switch to just using a docker catalina.
I'm taking AppleIDUUID and search-party-token from my normal mac and using them as docker environment variables. 
I'm trying to get X-Apple-I-MD and X-Apple-I-MD-M using https://github.com/Dadoum/Provision but the ones it generates always return a 401. Meanwhile using the ones that your request_reports.py generates, running it on a mac, works fine.

biemster, 2022-08-03:
The `X-Apple-I-MD-M` is tied to the machine ID, maybe the `search-party-token` is as well somehow? I did not try Dadoum/Provision yet, without a way to get a search-party-token it's not very useful to me.

olivluca, 2022-08-18:
To try to avoid the password prompt I reported in issue #1 I harcoded the AppleIDUUID and the search party token, however while the former appears to be constant, the latter varies from time to time, so I cannot hardcode it.

biemster, 2022-08-28:
I found a way to retrieve that AppleID UUID without accesssing the keychain, I only need to figure out where to find the `search-party-token`. But that seems to be the trickiest one.