biemster/FindMy #12: ST17H66 flashing

webfrank, 2022-11-05:
Hi, I have a 1$ aliexpress tag but don't understand how to flash openhaystack firmware. 

Any help?

biemster, 2022-11-06:
I indeed did not yet get to writing the documentation for this, could you post a picture of the board? There are three versions of the board that I know of, and they have different points on the PCB where you should connect the RX and TX of your UART to. GND and 3v should have obvious testpads.

1. UART RX should be connected to pin 6 of the 17h66, TX to pin 5.
2. Then start the flasher from this repo without connecting 3v to the chip, and when you hear the buzzer ticking do connect the 3v.
3. That should be all, the flasher should recognize the chip and flash your key in a few seconds.

If the flasher makes the buzzer tick, but does not recognize the chip after powering it up, you might have RX and TX mixed as I had in the beginning (not even sure now if I wrote them correctly here).
Also keep in mind that some usb-uart dongles can't deliver enough power to flash the chip, I have a PL2303 that did not work, but my CP2102 worked.

Hope this helps.

biemster, 2022-11-09:
@webfrank did you manage?

closed 2022-11-12 by biemster

webfrank, 2022-11-13:
Sorry for late reply. I found I have s ST17H26 and not 66. I think it cannot be flashed same way.

biemster, 2022-11-13:
The 26 is indeed a completely different animal. I have a few of those as well, and am still working on flashing those but I'm not very confident it will work since there are a lot of indications it is OTP only :(
Anyway if it turns out to be flash branded as OTP for marketing reasons, it will still need a completely different flasher as they are likely telink 8266 clones.

drott, 2022-11-16:
Looks like I got a batch of ...26 ones as well, and then another kind, looking different, labeled like this. Will try with a new order. Looks easier to solder if anything, but I have no idea about the chip. 

[image]

biemster, 2022-11-16:
This board I never saw before @drott , interesting! The silkscreen mentions june '22, so I doubt it's a 17h26 since they are superseded by the 66. But I might be wrong.
There is also no SWS pad on the board, but there are Tx and Rx! Did you try to see if they are alive?
Also the value on the oscillator will help identifying this filed down chip.

johnbaker26222, 2024-03-15:
I'm finding that even ordering from my same supplier the identical order, they are shipping with either ST17H66T which aren't flashable or a chip that I suspect is counterfeit. It's missing the ST17H66 and only has the B at the end. Both seem to not be flashable

isibizi, 2024-04-07:
Damn, i have the ST17H66T chip also in my tags. New news about that?

Systm21, 2024-04-08:
I have the same problem here... I got a new delivery of Lenze tags and there are a few with the model number ST17H66T. I haven't tested them yet, but if I go by the available datasheet, they should be easy to program serially via UART. 

But what's even weirder is that apart from the "T" models, I have received many more without a laser marking. The PCB layout is actually the same, but there is nothing on the chips. Has anyone had this before? They are completely unlabeled. 

Translated with DeepL.com (free version)

Systm21, 2024-04-08:
I have the same problem here... I got a new delivery of Lenze tags and there are a few with the model number ST17H66T. I haven't tested them yet, but if I go by the available datasheet, they should be easy to program serially via UART. 

But what's even weirder is that apart from the "T" models, I have received many more without a laser marking. The PCB layout is actually the same, but there is nothing on the chips. Has anyone had this before? They are completely unlabeled.

johnbaker26222, 2024-04-08:
Please let us know if you successfully flash an ST17H66T.

-------- Original Message --------

Systm21, 2024-04-08:
I haven't tried flashing a T chip yet, but have you noticed that the pinout is different in parts? P9 goes to VBUS, so that can't work. 

[image]

johnbaker26222, 2024-04-08:
Yes this is what I believe is right? 
[image]