biemster/FindMy #13: Flashing unidentified chip PCB label: RB-WS8031-FD01_V1.0

drott, 2022-11-16:
Spun off from #12. New photo, the crystal seems to say HY 16.000D.

[image]

When powering via VDD and GND, the TXD and RXD both go high, but I don't see any signal. Connecting with 9200 and 115200 baud and sending some keystrokes does not lead to anything so far (but I might have made mistakes in the analysis.)

drott, 2022-11-16:
These boards were in the black tags from https://www.aliexpress.com/item/1005003088294017.html

biemster, 2022-11-16:
My Lenze 17H66 have a 16MHz crystal as well, and my 17H26 boards have a 12MHz crystal. So we can already scratch off them being a 26, which is good!

biemster, 2022-11-16:
Is there anything on the other side?

drott, 2022-11-16:
> Is there anything on the other side?

No components, only some vias visible and the battery slot.

biemster, 2022-11-17:
What pin is the antenna on? And pin 5 in GND right?

drott, 2022-11-18:
4 and 5 have a ground connection, and connect to the antenna, yes (if that makes sense)

biemster, 2023-04-04:
I got those buggers in a batch of mine now as well :( At least that makes the investigation a bit more hands on. I hope the factory that produces these did not fully move away from the st17h66.

biemster, 2023-04-04:
At least google now knows what chip it is!
https://www.taoic.com/company/7287/WS8030%252FWS8031==3c2b0951-2bd8-11ec-9475-00163e1552d4-28_detail.html