biemster/FindMy #14: Low cost Apple-Airtag clones

Cyl0nius, 2022-12-07:
Apple-Airtag clone with ST17H66 on Aliexpress (https://www.aliexpress.com/item/1005004495296995.html)

Runs with Open-Haystack-App with no problems.

The minimal assembly of the pcb is interesting. No capacitors at the xtal and no antenna matching at all (see schematic).

Flashed with STC Auto-Programmer (CH340) without any problem.
1. connect P10 to TXD, P9 to RXD, GND to GND
2. start flash script
3. connect 3.3V to Battery +

All connection points are exposed on PCB.

[image]

[image]

[image]

[image]

biemster, 2022-12-07:
Nice find! Following the link I noticed that I put it on my wishlist as well already a while ago, but never ordered it :)
Great that you took out the solder iron and started slashing to a minimum, does that mean there was an LED and a switch originally?
And what is the fifth test pad connected to?

Cyl0nius, 2022-12-22:
More cheap tags available ...

[image]

[image]

[image]

biemster, 2022-12-22:
Nice :+1: Those I have as well, in fact they were the first vadimkozhin managed to flash :1st_place_medal:

Cyl0nius, 2022-12-30:
For easy programming the tags, even in bulk ... programming jig for 1mm pogo-pins.

STL is not supported, so I attach it as ZIP.

[image]

[st17h66_programming_jig.zip](https://github.com/biemster/FindMy/files/10323270/st17h66_programming_jig.zip)

drott, 2023-01-01:
> For easy programming the tags (button type as seen in first picture), even in bulk ... programming jig for 1mm pogo-pins.

Nice jig, thanks for sharing that. What are B and the button connected to? Do you need to keep the button pressed on power on or what do you do with it?

Cyl0nius, 2023-01-02:
B = black = GND
The button is simply a switch for power supply. After(!) starting the flash-script just press the button and hold it until flash is done.
I'm running the flash-script on Windows10.
Just run CMD as administrator (otherwise USB communication will not work) and execute the script with:
python3 flash_st17h66.py
After(!) starting the flash-script wait for a second, connect power (3VCD) till end of flash.

biemster, 2023-01-02:
> So far I failed to flash ST17H66B and C tags that I have here.

@drott How do you power your tags while flashing? They require more than some usb uarts can provide, as you might have already read in issue #5. Where does the flash script fail?

drott, 2023-01-02:
> @drott How do you power your tags while flashing? They require more than some usb uarts can provide, as you might have already read in issue #5. Where does the flash script fail?

I do connect a separate 3.3V power supply (based on a RD DPS5005 and a Makita battery) to the 3v3 pad or clamped the battery + metal. The flash script starts running, first I hear the buzzer with less volume, then with more when I connect the power, but I don't get much serial data back from the to-be-flashed tag: Occassionally I see some '\x00" serial responses printed, but not getting to the required `cmd>>` (or similar). I know ordered some SOP16 programming testing clips to see if I can make it work with those.

biemster, 2023-01-02:
Sorry I have to ask to eliminate, did you try swapping RX and TX? (I had them wrong way round for quite a while)

Cyl0nius, 2023-01-02:
I just want to mention it. 

The chip must be complete powerless (no battery connected) when the script starts.
Supply power only after a few seconds after the start of the flashing script.

aproxtimedev, 2023-01-05:
Sorry, i forgot to check this repo on folder Lenze_ST17H66

ggaljoen, 2023-02-21:
New pcb version, works perfect! [image]

Pogopins do their job fine: [image]

Added a switch for the power to the board.

olivluca, 2023-02-21:
@ggaljoen  nice, but from the photo I'm not sure I understand which pogo pin touches which point (red, yellow and orange should be on the same line, yet one of them is slightly displaced), please tell me if my interpretation in the following photo is correct. 
Also, how do you keep the jig aligned?

[image]

ggaljoen, 2023-02-21:
> Also, how do you keep the jig aligned?

Alignment is free hand practice, like this: 
[image]

Contact with P9 and 3V3 Vcc as guideline; 
[image]

biemster, 2023-03-03:
I finally received my tags like in the first post of this issue, they look really nice! By far the best of all the versions I have, tiny and tight pcb. Let's see if I manage this weekend to flash them, did anybody here try those airtag cases / holders / keychain things on these?

Pasarel, 2023-03-05:
Thank you Cyl0nius for the instructions, and biemster and vadimkozhin for the files in this repo. I flashed three devices like the ones in the first post with no issues, and they seem to be working fine atm.

Regarding those keychains sold on Aliexpress, I would advise against it. I bought one with my tags and it's too tight -- the tags don't fit. And looking through the reviews on Aliexpress, I am not the only person who faced this problem.

biemster, 2023-03-05:
> Regarding those keychains sold on Aliexpress, I would advise against it. I bought one with my tags and it's too tight -- the tags don't fit. And looking through the reviews on Aliexpress, I am not the only person who faced this problem.

Thanks for reporting this. Is it the tags being a different size than the original airtags, or are the keychains just not good for either?

Pasarel, 2023-03-05:
I haven't compared the size to the original yet, but I will in the next few days and I'll write back.

big-mak, 2023-05-02:
> I haven't compared the size to the original yet, but I will in the next few days and I'll write back.

Any update? Did a proof of concept on one like @Cyl0nius has, now looking to buy a bunch of airtag lookalikes.

vadimkozhin, 2023-06-10:
Just to confirm, that [this](https://www.aliexpress.com/item/1005004495296995.html) item (which was mentioned by @Cyl0nius) works. It has an original airtag formfactor and should be compatible with airtag accesories. The board have marked pads for TX and RX pins, so soldering will be a lot easier.

gammadog808, 2023-07-04:
Hi guys,
   I'm trying to flash a ST17H66(in windows), looks to be the same from the first post.  When I run the flash program, wait a few seconds and apply 3v3 to the board, I get this error in python:

Traceback (most recent call last):
  File "C:\dit\FindMy-monterey\Lenze_ST17H66\flash_st17h66.py", line 80, in <module>
    res = uart.read(10)
  File "C:\dit\venv\lib\site-packages\serial\serialwin32.py", line 295, in read
    raise SerialException("GetOverlappedResult failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: GetOverlappedResult failed (PermissionError(13, 'Access is denied.', None, 5))

Not sure why I get this error - I'm logged in as thee administrator, ran pycharm(or cmd) as admin, but no dice.  Even moved the default COM port on the usb flasher, still no luck.

Modified the COM line in the .py:
#uart = serial.Serial('/dev/ttyUSB0', 9600, timeout=0.01, inter_byte_timeout=0.01)
uart = serial.Serial('COM6', 9600, timeout=0.01, inter_byte_timeout=0.01)

Any ideas?

biemster, 2023-07-04:
@gammadog808 this seems to be an issue with serial on your machine indeed, and not with the script. Please try to get that running first with a simple test program (connect tx directly to rx, and see if it echos the characters)
Also, please open a separate issue if you continue having problems.

SergeKZ, 2023-07-04:
tell me where you can find the source Lenze_ST17H66

danhuanggt, 2023-09-20:
> I do connect a separate 3.3V power supply (based on a RD DPS5005 and a Makita battery) to the 3v3 pad or clamped the battery + metal. The flash script starts running, first I hear the buzzer with [...]

@drott Did you ever get your tags flashed? I'm encountering the same issue of not reaching the `cmd>>` and getting `\x00` responses back. Your experience mimics mine!

Here is how my device is hooked up to a CP2102.

[image]

humpataa, 2023-11-08:
Maybe I should have opened another "issue" but it somehow belongs here, so ...
Do these chinese AirTag clones actually work like Apple's AirTags?
I have tried to change the manufacturer bytes of Apple (0x004c) to 0x0501 (cheap chinese iTag). But this results in the tag not being reported to Apple's server anymore. At least requested reports no longer include info for it anymore. I have not changed anything else, just the manufacturer bytes.
I can see using nRFConnect that the original iTag is using a much shorter advertising string – 15 bytes instead 30 bytes ...
Has anyone looked into this? How are they "using" Apple's network?
Or do I have to change the request for reports to get info about tags with different manufacturers?
Any hint is appreciated.

biemster, 2023-11-08:
@humpataa This should go to another issue indeed, where you might explain your question in a bit more detail. Changing the manufacturer bytes would obviously cause the tag not to be reported anymore. Cheap chinese iTags do not participate in Apple's FindMy network (but it is possible I don't understand your question)

humpataa, 2023-11-08:
Thanks for the quick reply. So what network do they use? Android, Samsung, Google – I didn't know that there actually IS someone else doing this like Apple does ...

biemster, 2023-11-08:
those cheap iTags don't use any distributed network, the registered phone just remembers where it saw them last.

humpataa, 2023-11-08:
oh really? that really sounds like "chinese" ... and makes them rather useless if not flashed. thank you!

biemster, 2023-11-08:
Apple's FindMy network seems to be the only current solution, besides smaller networks that require dedicated apps like Tile or Chipolo. Although Android might join the game soon: https://www.zdnet.com/article/is-googles-find-my-device-network-for-android-nearing-settings-signs-point-to-yes/

steve-m, 2024-01-05:
Unfortunately the tags from the first post are not available anymore. I ordered similar looking ones, but unfortunately they have the ST17H66T variant with OTP ROM, so they are useless. Does anyone have a source for tags with the B2 flash variant?

isibizi, 2024-04-07:
> Unfortunately the tags from the first post are not available anymore. I ordered similar looking ones, but unfortunately they have the ST17H66T variant with OTP ROM, so they are useless. Does anyone have a source for tags with the B2 flash variant?

Are you sure that the T variant not working with script? @Cyl0nius do you have any news about this issue?

zjonesz, 2024-09-17:
[image]

liufuhu, 2024-10-05:
sh17h66T chip can not flash，66t is OTP chip，66b can relfash

Systm21, 2024-11-12:
I asked Aaron if he had any ideas for a cheap but economical OHS transmitter. Aaron hacked Bluetooth and other devices and was actually able to help. He has written his own webflasher for the 1.50€ TB-03F. 

Have a look for yourself and support him, there is a lot to discover with him, such as the game Doom on a toothbrush. 

https://x.com/atc1441/status/1856136207501037916

josemariaaraujo, 2024-11-15:
Hi, I've ordered a bunch (5 different styles/vendors) of low cost(<2€) bluetooth trackers from aliexpress hoping I could find one with ST17H66B, but all but one came with an umarked SOT8 that I suppose is the WS8300 (OTP) and the other one came with ST17H66T, also OTP.

Does anyone have a good source for a low cost tracker already with plastic case with battery compartment that can be reprogrammable to join the FindMy network?

Thank you.

omarkhali, 2024-11-30:
> I asked Aaron if he had any ideas for a cheap but economical OHS transmitter. Aaron hacked Bluetooth and other devices and was actually able to help. He has written his own webflasher for the 1.50€ TB-03F.
> 
> Have a look for yourself and support him, there is a lot to discover with him, such as the game Doom on a toothbrush or OpenEpaperLink.
> 
> https://x.com/atc1441/status/1856136207501037916

Hello I try to flash firmware to Mi LYWSD03MMC,TB-03F  but there are no advertise of public key

biemster, 2024-11-30:
> Hello I try to flash firmware to Mi LYWSD03MMC,TB-03F but there are no advertise of public key

Any other signs of life from the MCU? Did you flash this chip before?
I'll need some more info to be able to help, also please open a new issue for this.

lovelyelfpop, 2024-12-01:
I was able to flash the hex file, but the tag not working, bluetooth seactch cannot found it, FindMy app cannot discover it. There is no response when I press the button on the board.

[image]

This is my board with a chip ST17h66B
[image]

biemster, 2024-12-01:
@lovelyelfpop  I'm going to ask you the same as omarkhali in literally the first comment above yours, please open a new ticket instead of overtaking someone else's.

lovelyelfpop, 2024-12-02:
Sorry, it did work, I can see locations on android app.  The button on the board is useless after flash the firmware