biemster/FindMy #17: "invalid padding bytes"

noobplayer480, 2023-01-04:
I m using big sur11.0.1 on VMware, iCloud working properly.

[image]

biemster, 2023-01-04:
you should use your MacBook login password, odd that it returns no data then. do you have multiple icloud accounts on that vm?

noobplayer480, 2023-01-04:
@biemster no sir , only one , i don't have any iPhone in my area , do u think it's bcz it doesn't have any locations reports ? 
I got succes with openhaystack but map didn't work , black screen in map bcz of VM. [image]

closed 2023-01-04 by biemster