biemster/FindMy #18: Cannot generate keys.

mishailovic, 2023-02-18:
```
hotaru@mbp-hotaru FindMy % python3 generate_keys.py
Traceback (most recent call last):
  File "/Users/hotaru/FindMy/generate_keys.py", line 33, in <module>
    priv_bytes = int_to_bytes(priv, 28)
  File "/Users/hotaru/FindMy/generate_keys.py", line 9, in int_to_bytes
    s = ('0'*(len(h) % 2) + h).zfill(length*2).decode('hex')
AttributeError: 'str' object has no attribute 'decode'
```

biemster, 2023-02-18:
Did you checkout the main branch or the monterey branch? Main should be run with python2, monterey with python3. The shebang takes care of that, just run `$ ./generate_keys.py`

mishailovic, 2023-02-19:
> Did you checkout the main branch or the monterey branch? Main should be run with python2, monterey with python3. The shebang takes care of that, just run `$ ./generate_keys.py`

Yes, of course

mishailovic, 2023-02-19:
Not really the question about this issue, but, is there any way I could export my devices from openhaystack to the .keys format for this python script?

biemster, 2023-02-19:
> > Did you checkout the main branch or the monterey branch? Main should be run with python2, monterey with python3. The shebang takes care of that, just run `$ ./generate_keys.py`
> 
> Yes, of course

Can you give me some more info on your system; OS,python version and which branch you checked out for example. This looks like a general python issue.

biemster, 2023-02-19:
> Not really the question about this issue, but, is there any way I could export my devices from openhaystack to the .keys format for this python script?

I never used openhaystack directly (no Mac), but if it allows to export both the public and private keys than it would be easy to copy them in a .keys file.

mishailovic, 2023-02-19:
> Can you give me some more info on your system

Sure, I am using MacOS Monterey 12.6.3, my python version is Python 3.9.6, I checked out monterey branch before running code

biemster, 2023-02-19:
The private key B64 is the number you get on [line 36](https://github.com/biemster/FindMy/blob/monterey/generate_keys.py#L36), combining with the couple lines before and after that you should be able to generate a .keys file from that. Let me know if you get stuck.

biemster, 2023-02-19:
I'm getting the same error as you actually now trying it on Monterey, I guess I never tested this properly.

mishailovic, 2023-02-19:
> I'm getting the same error as you actually now trying it on Monterey, I guess I never tested this properly.

Yeah, I kinda think the whole thing should be rewritten to the python3 cuz python2 is dead as hell 💀 I couldn't even install packages via pip on it, just throws me ssl exceptions because of expires certificates

biemster, 2023-02-19:
That's what I did on the monterey branch, but I kept it on python2 because that is installed by default on Catalina. Monterey doesn't even come with python preinstalled anymore, and I made this repo for me (and others) who run macOS in a VM solely to run this script. Then Catalina + python2 is the easiest in my opinion, no additional requirements.
But yeah, python2 should be phased out.

mishailovic, 2023-02-19:
thx, issue fixed

closed 2023-02-19 by mishailovic

johnbaker26222, 2023-05-31:
Sorry so the hashed adv key that this script generates isn't necessary to change if you only use the openhaystack app? So just copy public and private key to the .keys file and flash that?

biemster, 2023-05-31:
If you only use the openhaystack app you don't need a .keys file right? I don't think I understand the question, what part of this repo do you want to use?

johnbaker26222, 2023-05-31:
Sorry I didn’t ask very clearly. Is there a way to use the same key for both the openhaystack app and also when using requests script as well? Can I get the hashed adv key for the request via OHS app?

biemster, 2023-05-31:
There seems to be a "Copy advertisement key" option in openhaystack: https://github.com/biemster/FindMy/issues/18#issuecomment-1435952938, but I don't know what it exactly returns (I never used the app). What for sure works is copy the private key b64 and plug that in the `generate_keys.py` script, as described in https://github.com/biemster/FindMy/issues/18#issuecomment-1435990649.

After that you can use that same key in both openhaystack and the request script.

johnbaker26222, 2023-05-31:
Put it in line 36? Could you show an example of how that would be placed in the code?

biemster, 2023-05-31:
https://github.com/biemster/FindMy/blob/d122192999c69d36975ae1d356e3ac5c3281fa3c/generate_keys.py#L28

should be changed to
```python
priv = bytes_to_int(base64.b64decode('<your priv B64 here>'))
```

johnbaker26222, 2023-05-31:
I'm getting this:
`NameError: name 'bytes_to_int' is not defined

biemster, 2023-05-31:
ah yes, are you on python2 or 3?

johnbaker26222, 2023-05-31:
I’ve got your Monterey version checked out so 3

biemster, 2023-05-31:
the function lives here: https://github.com/biemster/FindMy/blob/d122192999c69d36975ae1d356e3ac5c3281fa3c/request_reports.py#L17