biemster/FindMy #19: No data returns

davesenior9, 2023-03-14:
Im using the FindMy_proxy and client on a established and working OHS Setup. Tags are coming in as expected. When I use the FindMy_client.py script it connects to the proxy fine, but returns 0 reports received, found: [], missing [].

Any idea where to start?

biemster, 2023-03-14:
It seems you don't have any `.keys` files in your working directory. Did you generate any with `generate_keys.py` and deploy? Or are you trying to request keys that are deployed with OHS?

davesenior9, 2023-03-15:
Trying to request keys that are deployed with OHS. All deployed devices were deployed organically within OHS.

biemster, 2023-03-15:
To generate `.keys` files from OHS deployed devices you can have a look at the discussion here: https://github.com/biemster/FindMy/issues/18#issuecomment-1435990649
Let me know if you run into issues.

davesenior9, 2023-03-16:
awesome. good to go. thanks :)

biemster, 2023-03-16:
no problem!

closed 2023-03-16 by biemster