biemster/FindMy #20: 400 Bad Request

davesenior9, 2023-03-30:
```
400 Bad Request
Traceback (most recent call last):
  File "request_reports.py", line 189, in <module>
    res = json.loads(response.read())['results']
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
```

Have you encountered this before? I have automated the polling of this script, and had it running for a few days. It abruptly stopped and generated this error. Restarting the script made no change. If i go back to vanilla OHS and use the GUI, it updates fine.

Thanks!

EbonyHuang, 2023-03-30:
We see the same issue here. If you print out response.status and response.reason, it shows that 400 Bad Request. This seems mean the incorrect header/data is posted to server. 
But why it happened?? Not sure what's wrong???

biemster, 2023-03-30:
Odd stuff! What OS are you on @davesenior9 and @EbonyHuang ? I'll have to spin up my macOS dockers to see if I can reproduce. Do both the `request_reports.py` and `FindMy_client.py` fail for you? (it might be related to the `startDate` and `endDate` fields, I've noticed some server-side changes in that area).

EbonyHuang, 2023-03-30:
For both BigSur and Monterey OS, the request_report.py has the same failure. It all works couple days ago. Thank you.

EbonyHuang, 2023-03-30:
Looks like it is the https body(search) contain the single quote ' in the "ids" list, when we replace that single quote to be double ". It seems work again. 
add this line before do the https post.
data = data.replace("'",'"')

davesenior9, 2023-03-31:
awesome work. thats fixed it for me. I would be interested to know what caused this. Some sort of change on the Apple gateway?

biemster, 2023-03-31:
Well done @EbonyHuang ! The `ids` list was a bit hacky indeed, I guess Apple updated a json parser somewhere, not allowing single quotes anymore. I'll put in a fix when I find the time.

biemster, 2023-04-26:
Reopening because this definitely needs the fix by @EbonyHuang

biemster, 2023-11-01:
This should be fixed now in both the catalina and monterey branches. Also main of course but that is undergoing major overhaul at the moment, to abolish the requirement of a mac altogether.

closed 2023-11-01 by biemster