biemster/FindMy #23: Getting Permission errror, Access is denied when flashing ST17H66

gammadog808, 2023-07-04:
Hi guys,
As requested by Biemster, I made my own thread on this.

I'm trying to flash a ST17H66(in windows), looks to be the same from the first post. When I run the flash program, wait a few seconds and apply 3v3 to the board, I get this error in python:

Traceback (most recent call last):
File "C:\dit\FindMy-monterey\Lenze_ST17H66\flash_st17h66.py", line 80, in
res = uart.read(10)
File "C:\dit\venv\lib\site-packages\serial\serialwin32.py", line 295, in read
raise SerialException("GetOverlappedResult failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: GetOverlappedResult failed (PermissionError(13, 'Access is denied.', None, 5))

Not sure why I get this error - I'm logged in as thee administrator, ran pycharm(or cmd) as admin, but no dice. Even moved the default COM port on the usb flasher, still no luck.

Modified the COM line in the .py:
#uart = serial.Serial('/dev/ttyUSB0', 9600, timeout=0.01, inter_byte_timeout=0.01)
uart = serial.Serial('COM6', 9600, timeout=0.01, inter_byte_timeout=0.01)

BTW, I'm using the Monterey branch and newer 3.9/3.10.3.11 versions of python.

Any ideas?

gammadog808, 2023-07-04:
I should also add that I have hooked up putty to the device and it doesn't complain of port being used, so that's not the issue. 

Also, when connecting to putty and I connect the TXD and RXD, I do get funky characters but it pumps something whenever I retouch the pads.  However, as soon as I connect power, putty kicks me out with an error along the lines with you can't do that.

So, maybe it's a windows setting, etc?  If so, I'm unaware what it could be - I looked in device manager and messed with a few settings, but same output.

gammadog808, 2023-07-05:
Also wanted to mention, for ground and power, I'm using the connectors on the bottom vs the pads on the top. 

Don't think it matters, but figured I'd throw it out there.  I did notice that if I just apply power to the board with TXD and RXD disconnected, putty will give me the same connection error.  Maybe it's the board?

Power and ground is coming from the usb flasher.

gammadog808, 2023-07-05:
I think I got a little further.  I saw someone saying they used the 5v connection.  Since I have a few to work with, I gave it a shot. With putty and the program, I don't get the access problem.  

Now, my question is what output should I see/expect to happen?  Does anyone have a sample output or tell me what I should expect?  I see the .py script do a few things then end with a clean exit.  But I don't see a .key file, etc generated?

biemster, 2023-07-05:
The output should have a lot of #OK lines, so if you are not sure it's working, it's definitely not. It's possible that your USB UART does not provide enough power to the st17h66 to flash it, what's the chip on your dongle?

gammadog808, 2023-07-05:
Biemster,
  These are the flashers I'm using, basically a CH340 chip:
[https://www.amazon.com/HiLetgo-Module-Microcontroller-Download-Serial/dp/B00LZV1G6K/ref=sr_1_3?crid=15CG0CGYVSAAV&keywords=ch340&qid=1688589901&sprefix=ch34%2Caps%2C366&sr=8-3](url)

Here's the output of one without the adv key.  Strangely, I had to keep connecting the power from the tag to move from line to line:
>python flash_st17h66.py
Usage: flash_st17h66.py <pubkey base64>
Using default key instead
Start of new ihex section found, assuming ER_ROM_XIP
Start of new ihex section found, assuming JUMP_TABLE
Start of new ihex section found, assuming ER_IROM1
b'\x03\x00\x00\x00\xff\xff\xff\xff8\x18' ... b'\x00\x008\x18\xff\x1f\xff\xff\xff\xff', len 304 sum 67861
b'\x88p' ... b'\x0c\x04' ... b'\x0c;'
b'\x00\x00\x00\x00\x01E\xff\x1f Q' ... b'E^\x02\x11\xe8\x01\x18\x00\x00\x00', len 16160 sum 1385258
b'\x03\xb4\x01H\x01\x90\x01\xbd\xad\xae' ... b'\x01\x01\x01\x01\x03\x03\x02\x01\x03\x00', len 28808 sum 2778341
key: b'\x11"3DUfw\x88\x99\xaa\xbb\xcc\xdd\xef\xfe\xdd\xcc\xbb\xaa\x99\x88wfUD3"\x11'
b'\x00'
b'cmd>>:'
RESET MODE activated. Changing baudrate to 115200
Erase + Write
sent b'er512'
Response is: b'#OK>>:'
sent b'rdrev+'
Response is: b'0x0012600b 6222M005 #OK>>:'
sent b'wrreg4000c890 ab000001 '
Response is: b'#OK>>:'
sent b'wrreg4000c838 ff010005 '
Response is: b'#OK>>:'
sent b'spifs 0 1 3 0 '
Response is: b'#OK>>:'
sent b'sfmod 2 2 '
Response is: b'#OK>>:'
sent b'cpnum ffffffff '
Response is: b'#OK>>:'
sent b'cpbin c0 002000 130 11002000'
Response is: b'by hex mode:'
sent c0 (len=304)
Response is: b'checksum is: 0x00010915'
sent checksum
Response is: b'#OK>>:'
sent b'cpbin c1 005000 3f20 11005000'
Response is: b'by hex mode:'
sent c1 (len=16160)
Response is: b'checksum is: 0x0015232a'
sent checksum
Response is: b'#OK>>:'
sent b'cpbin c2 020000 7088 11020000'
Response is: b'by hex mode:'
sent c2 (len=28808)
Response is: b'checksum is: 0x002a64e5'
sent checksum
Response is: b'#OK>>:'

Looks like if I don't put a key in there, a default one is burnt for me.  Since then, I did make a device on openhaystack, grab it's adv key and pass it in to flash_st17h66.py and marked it as deployed.  Assuming I did that all right, I'd assume I'd see that tag come online, but nope.

biemster, 2023-07-05:
That output looks right, so that's working! Do you have a bluetooth app like nRF Connect or something to check the advertisements around you?
Also, what's the error when you try the adv key generated by the script?

gammadog808, 2023-07-05:
I don't, but I'll see if I can get one.  As for the error, I don't have it - I think it was something along the line of the string is not valid, then it would default to the default key.  Let me plug away a bit and see if I can gather more info.  Thank you for the help!

biemster, 2023-07-05:
nRF Connect is just an app on your phone, very useful for this kind of stuff.
If the adv string is not valid, did you but the base64 string of the raw bytes? could you try with a different key?

gammadog808, 2023-07-05:
Biemster,
   Yup, I just pulled one off the app store.  I see a ton of bluetooth devices(in the office) so it's hard for me to distinguish which might the tag.
W
e got further, sorry the, "bad" key was because I was rushing and grabbed the hashed adv key.  Now it seems ok.  Now we run the report, but don't see it.  So we're troubleshooting.  To be clear, once we flash the tag, we don't need to register it at all in icloud/phone, etc, correct?  Just run request_reports.py with the appropriate key?

closed 2023-07-20 by biemster

danhuanggt, 2023-09-19:
@gammadog808 I am also using the same CH340 flasher and I might be encountering the same issues you had.

Do mind sharing which wires you connected on your CH340 to which pads on the ST17H66?

This is the output from terminal:
> python3 flash_st17h66.py
> Usage: flash_st17h66.py <pubkey base64>
> Using default key instead
> Start of new ihex section found, assuming ER_ROM_XIP
> Start of new ihex section found, assuming JUMP_TABLE
> Start of new ihex section found, assuming ER_IROM1
> b'\x03\x00\x00\x00\xff\xff\xff\xff8\x18' ... b'\x00\x008\x18\xff\x1f\xff\xff\xff\xff', len 304 sum 67861
> b'\x88p' ... b'\x0c\x04' ... b'\x0c;'
> b'\x00\x00\x00\x00\x01E\xff\x1f Q' ... b'E^\x02\x11\xe8\x01\x18\x00\x00\x00', len 16160 sum 1385258
> b'\x03\xb4\x01H\x01\x90\x01\xbd\xad\xae' ... b'\x01\x01\x01\x01\x03\x03\x02\x01\x03\x00', len 28808 sum 2778341
> key: b'\x11"3DUfw\x88\x99\xaa\xbb\xcc\xdd\xef\xfe\xdd\xcc\xbb\xaa\x99\x88wfUD3"\x11'
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py", line 575, in read
>     buf = os.read(self.fd, size - len(read))
>           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> OSError: [Errno 6] Device not configured
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/Users/.../.../.../FindMy/Lenze_ST17H66/flash_st17h66.py", line 79, in <module>
>     res = uart.read(10)
>           ^^^^^^^^^^^^^
>   File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py", line 581, in read
>     raise SerialException('read failed: {}'.format(e))
> serial.serialutil.SerialException: read failed: [Errno 6] Device not configured

biemster, 2023-09-19:
@danhuanggt does your Mac have a `/dev/ttyUSB0`?
https://github.com/biemster/FindMy/blob/ed375cb9758a98253007cb825d2b524e0d7b4fed/Lenze_ST17H66/flash_st17h66.py#L74

I kind of remember macOS using a different naming scheme for this, if so you should edit that line in the flasher.

danhuanggt, 2023-09-19:
> @danhuanggt does your Mac have a `/dev/ttyUSB0`?
> 
> https://github.com/biemster/FindMy/blob/ed375cb9758a98253007cb825d2b524e0d7b4fed/Lenze_ST17H66/flash_st17h66.py#L74
> 
> I kind of remember macOS using a different naming scheme for this, if so you should edit that line in the flasher.

I am using `/dev/tty.usbserial-1130`, which is edited in the `flash_st17h66.py` file.
Also with a jumper between **TXD** and **RXD** with a simple PySerial script, I am able to output and confirm that the flasher works, but when I hook it up to the ST17H66 and try and run the actual flash script is when the above error occurs.

@biemster do you think it could be my CH340 UART? I do have a CP2102 I could try. Were you successful in using that particular UART?

biemster, 2023-09-19:
It's definitely possible your CH340 is not able to provide enough power to flash the chip. When you put the st17h66 in programming mode it starts to consume several 100s milliamps for some reason, I had the same issue before with a PL2303: https://github.com/biemster/FindMy/issues/5#issuecomment-1201484148. I had better luck with a CP2102 after that, so yes please try with that one.

gammadog808, 2023-09-19:
Dan, it's been a while, but I do know I had to use the 5v output, the 3.3v never worked for me.

danhuanggt, 2023-09-20:
Update: So I've tried both the CH340 (running on 5v, thanks @gammadog808) and the CP2102 and I've gotten a bit further.

When I start the script I hear repeated light clicking, when I connect 3v3 shortly after, the clicking gets louder but the flashing stalls. Whenever I touch 3v3, it outputs `b'\x00'`.

This is the latest output:

> python3 flash_st17h66.py
> Usage: flash_st17h66.py <pubkey base64>
> Using default key instead
> Start of new ihex section found, assuming ER_ROM_XIP
> Start of new ihex section found, assuming JUMP_TABLE
> Start of new ihex section found, assuming ER_IROM1
> b'\x03\x00\x00\x00\xff\xff\xff\xff8\x18' ... b'\x00\x008\x18\xff\x1f\xff\xff\xff\xff', len 304 sum 67861
> b'\x88p' ... b'\x0c\x04' ... b'\x0c;'
> b'\x00\x00\x00\x00\x01E\xff\x1f Q' ... b'E^\x02\x11\xe8\x01\x18\x00\x00\x00', len 16160 sum 1385258
> b'\x03\xb4\x01H\x01\x90\x01\xbd\xad\xae' ... b'\x01\x01\x01\x01\x03\x03\x02\x01\x03\x00', len 28808 sum 2778341
> key: b'\x11"3DUfw\x88\x99\xaa\xbb\xcc\xdd\xef\xfe\xdd\xcc\xbb\xaa\x99\x88wfUD3"\x11'
> b'\x00' (everytime I reconnect 3v3)
> b'\x00'
> ...

biemster, 2023-09-20:
Did you try it with tx and rx the other way around? Also for me it usually doesn't work on the first try, just cycle the power a few times. I have a loose dupont cable on the 3v3, press it in the breadboard to power the chip, and if I don't get `cmd>>:` I just take the dupont cable of and immediately try again. After a few tries it should work otherwise reverse tx and rx and repeat.

danhuanggt, 2023-09-20:
Okay @biemster I will attempt that next. Could it also be that my IC chip isn't actually a ST17H66? The labeling is sanded off, but it matches a lot of the diagrams I see online. It also has a button, which I'm not sure if that makes any difference.

Here is how I am currently connected via the CP2102:
3v3 -> 3.3v+
GND -> GND
TXD -> P9
RXD -> P10

[image]

My PCB in question:
[image]
[image]

I'm a software guy by trade but hardware was my first love, so I apologize for all the clarifying questions -- Let me know if I should start another issue instead of continuing in this thread.

biemster, 2023-09-20:
Yeah that's not a st17h66, but a clone. I got a few tags with them, and @agittins managed to actually flash one of them (https://github.com/biemster/st17h66_RF/issues/8), but I did not have time to try out mine yet. Good idea to follow up on that issue there indeed, and leave this closed.
Please do post any further findings you have in that issue on the st17h66_RF repo, because it seems more and more tags have this chip so it would be nice to get it working! I'm going to actively research on this as well there.
thanks!

biemster, 2023-09-21:
just to close  this side discussion, it was the tx and rx reversed (https://github.com/biemster/st17h66_RF/issues/8#issuecomment-1728567761). Happens to me all the time as well.