biemster/FindMy #24: Anyone know the legality/ethics of this?

gammadog808, 2023-07-06:
Hi guys,
   Wondering if anyone had input on the legality/ethics of this.  We got it working here, and it's great!  It would *so* useful for an application here at work.  Way more updates than what you get via airtags(due to anti-stalking mode, I assume).  But as it would be utilized in a work environment vs a home, "for fun/learning" environment, I wasn't sure how to approach this.

Part of me thinks it's an open BLE standard, so we just jump on whoever/whatever supports it...which in this case is apple's(only?) network.  Another part of me thinks it's like taking something from an unlocked car.  Just because I can get in doesn't mean it's ok(owner didn't give me permission).  On the positive side, you would/could buy apple hardware to support the server side, so there is something in it for apple as well.  Wondering if anyone had any ideas on this.

Cyl0nius, 2023-07-16:
No one is forcing you to take things out of an open car or throw in an airtag. 
If you have problems with this project, just move on, nothing to see here.

gammadog808, 2023-07-16:
I think you are misunderstanding me.  I am soooo stoked about this project!  At work, we are having a blast working with it. 

However, if we want to deploy and use it in an internal project for work, we want to make sure we do it the right way.  I’ve since been trying to get in touch with Apple to get clarification, but I haven’t gotten anything definitive.  I’m really hoping we can use it as it will really help our first responders here.

biemster, 2023-07-20:
Very interesting question, but I'm going to delete this issue.

closed 2023-07-20 by biemster