biemster/FindMy #25: Never receiving any reports?

Yannik, 2023-08-14:
Hi,
I have successfully created the `.keys` file with `generate_keys.py`. Now I'm trying to get reports using `request_reports.py`.
 (I have applied the `data = data.replace("'", '"')` fix).

For some reason, `request_reports.py` ALWAYS reports `0 reports received.`/`0 reports used.`. (Even though I have 2 devices and 10 items registered).

```
$ python3 request_reports.py 
Keychain password:
/Users/yannik/Library/Python/3.9/lib/python/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
200 OK
0 reports received.
0 reports used.
found:    []
missing:  ['BCBWDhj']
```

I'd greatly appreciate any hints how to fix this.

Yannik, 2023-08-14:
Same thing happens with `FindMy_proxy` and `FindMy_client`:

```
$ python3 FindMy_proxy.py
/Users/yannik/Library/Python/3.9/lib/python/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
Keychain password:
{"search": [{"startDate": 0, "ids": ['XXX/FYRC2M=']}]}
2023-08-14T10:46:34Z 127.0.0.1 200 {"search": [{"startDate": 0, "ids": ["XXX/FYRC2M="]}]}
{"search": [{"startDate": 0, "ids": ['XXX/FYRC2M=']}]}
2023-08-14T10:46:52Z 127.0.0.1 200 {"search": [{"startDate": 0, "ids": ["XXX/FYRC2M="]}]}
``` 

```
$ python3 FindMy_client.py
0 reports received.
found:    []
missing:  ['BCBWDhj']
$ python3 FindMy_client.py
0 reports received.
found:    []
missing:  ['BCBWDhj']
```

Cyl0nius, 2023-08-14:
Missing keychain password?
Are you successfully logged in to iCloud?

[image]

Yannik, 2023-08-15:
@Cyl0nius Yes, I have typed in the keychain password, and I'm successfully logged in to iCloud. Syncing the Keychain to iCloud was not enabled, but I have enabled it now. Unfortunately, this did not help. Still zero reports.

Just to be sure I understand the capabilities of this project correctly: I should be able to see the locations of regular airtags and apple devices, right? Or does it only work for self-flashed tags?

biemster, 2023-08-15:
The `200 OK` seems to indicate that you authenticated to icloud properly. your last question might indicate where the problem is, what kind of device did you flash with your generated key from your `.keys` file? or are you trying to use it with a regular airtag/idevice? It can work with those too, but you'll need to obtain the keys for those devices first, the `generate_keys.py` script cannot do that for you.

biemster, 2023-08-17:
Obtaining those keys involves getting into the secured storage of the official FindMy app, which needs system integrity protection (SIP) to be disabled. This is generally considered not such a good idea, and on top of that I have no idea how to do any of that.. :frowning_face: (I don't own any airtags or idevices)
There might be guides somewhere that can explain how to do this, but I never tried it, sorry.

Yannik, 2023-08-17:
Thanks @biemster 

If anybody else can chime in with expertise how to retrieve the keys (or a link to such information), I'd be very glad :-)

closed 2023-11-01 by biemster