biemster/FindMy #26: Tracker running on single 1.5V coin battery

Cyl0nius, 2023-08-28:
I am very happy with the ST17H66 ... except that the thing requires a (relatively) large 3 volt battery. 
In order to be able to track small objects (pen) or maybe even implanted, I need another BLE SOC.
The IN100 from Inplay could be the appropriate solution at a good price of under a dollar.

A CR2032 (20x3.2mm) has approximately a capacity of 200mAh.
A  ZA13 (7.8x5.3mm) has approximately a capacity of 300mAh.

https://inplay-tech.com/in100

biemster, 2023-10-19:
How did I miss this issue, that chip looks awesome! With a sleep current of about 0.5 uA and a 300mAh ZA13 you don't need a case, just epoxy them together since it will last for over a year!
Did you make any progress here?

Cyl0nius, 2023-11-03:
Yes ... now I'm looking for a way to make flex PCB myself.

[image]

biemster, 2023-11-03:
Nice! Do you have a target for the final cost of an assembled module? I guess it only needs an additional crystal and a couple passives? (hardware is not my forte..)
I'm actually looking a bit as well at alternative chips, the CH582 looks very interesting too, although it's a bit bigger and needs 1.8v.

davesenior9, 2023-12-19:
Did you end up building these ?