biemster/FindMy #28: 'status': 0 field?

jrusi, 2023-10-09:
First of all thx a lot for an amazing code. 
I'm not sure if this is the right place to ask:
What is the 'status': 0 field used for? Can we use it for indicating a button press on the beacon or some other sensor data like low battery level indication?

olivluca, 2023-10-09:
Please see https://github.com/biemster/st17h66_FindMy/issues/6

biemster, 2023-10-09:
Yes, it's a byte you control. I believe Dahknod's FakeTag uses it as a battery indicator (https://github.com/biemster/st17h66_FindMy/issues/6), but a button press is possible too (but keep in mind you might not get the notification for an hour or so, if at all)
Also according to https://github.com/dchristl/headless-haystack/issues/15#issuecomment-1732738154 it can be used to pretend to be an iPhone, and trick the anti-tracking saveguards.

biemster, 2023-10-09:
@olivluca beat me to the punch!

closed 2023-10-09 by biemster

olivluca, 2023-10-09:
I didn't know that leaving it at 0 will avoid the anti-stalking measures without the need of key rotation, so maybe it's better to leave it 0.

biemster, 2023-10-09:
It's a bit more subtle than that, key rotation will also not prevent iPhones from displaying an alert. But it will prevent tracking based on MAC address.
Setting the status to 0x00 apparently will trick an iPhone into thinking it is another iPhone, and not display an alert. But I did not test this yet.