biemster/FindMy #29: ValueError: Invalid padding bytes.

humpataa, 2023-10-22:
I am using monterey and python3 version. Can generate keys, looks good.
Whenever I run request_reports.py it asks me for keychain password, I use my Apple-id password (when I try to use my macbook password, it breaks with 400 Bad Request). I am also logged in via Chrome at Apple's iCloud website.
The script breaks with the above message.
Any idea?
Note: the generated key is not yet flashed to any device.
But I assume there should be some other message than just crashing?!
Any help is appreciated.

Systm21, 2023-10-22:
keychain password is your macbook password...

humpataa, 2023-10-22:
Thank you, now I get a 400 Bad Request followed by some more output ... 

```
Keychain password:
400 Bad Request
Traceback (most recent call last):
  File "/Users/humpataa/Library/Python/3.7/lib/python/site-packages/requests/models.py", line 971, in json
    return complexjson.loads(self.text, **kwargs)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "request_reports.py", line 179, in <module>
    res = response.json()['results']
  File "/Users/humpataa/Library/Python/3.7/lib/python/site-packages/requests/models.py", line 975, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

biemster, 2023-10-22:
You're probably hitting issue #20 ? There is an easy fix for that https://github.com/biemster/FindMy/issues/20#issuecomment-1490910909 and #22 , but I (still!) did not get to push that. You'll have to add this by hand for now I'm afraid.

humpataa, 2023-10-22:
Ah yes, I just found it myself. Thank you!

humpataa, 2024-07-29:
I am again getting the message "Invalid padding bytes" although I have solved it in the past.
Unfortunately, I don't have a clue now – the key is correct, password is the one of my Macbook ... the "mmeTokenFile" looks good, the "decryption_key" looks valid ... so what else could it be?

```
Traceback (most recent call last):
  File "/Users/humpataa/request_reports.py", line 186, in <module>
    AppleDSID,searchPartyToken = getAppleDSIDandSearchPartyToken(iCloud_decryptionkey)
  File "/Users/humpataa/request_reports.py", line 111, in getAppleDSIDandSearchPartyToken
    decryptedBinary = unpad(decrypt(open(mmeTokenFile, 'rb').read(), algorithms.AES(decryption_key), modes.CBC(b'\00' *16)), algorithms.AES.block_size)
  File "/Users/humpataa/request_reports.py", line 31, in unpad
    return unpadder.update(paddedBinary) + unpadder.finalize()
  File "/Users/humpataa/Library/Python/3.10/lib/python/site-packages/cryptography/hazmat/primitives/padding.py", line 159, in finalize
    result = _byte_unpadding_check(
  File "/Users/humpataa/Library/Python/3.10/lib/python/site-packages/cryptography/hazmat/primitives/padding.py", line 101, in _byte_unpadding_check
    raise ValueError("Invalid padding bytes.")
ValueError: Invalid padding bytes.
```
Any idea?

closed 2024-07-29 by humpataa