biemster/FindMy #33: how many keys max in one request?

humpataa, 2023-11-09:
just wondering ... have you ever noticed a limit how many keys one post can include?

biemster, 2023-11-09:
I've never tried this, but I've heard rumors about a limit of 256. Could you report back here after you tested this?

Chapoly1305, 2023-12-05:
@biemster @humpataa 
just tested, [1,96] are allowed. Will fail on 97th. The size of response seems irrelevant. 

I have tested to get 96 device keys at one request and received a 284,512 Bytes response that contains 96 keys' reports.

biemster, 2023-12-05:
Thanks @Chapoly1305 !

closed 2023-12-05 by biemster

humpataa, 2023-12-07:
Hmm, I've been requesting data for over 280 keys at once lately, once per hour, haven't seen a problem.
But as you seem to be sure ... I wasn't actually watching the response I must admit (pushing the response right into db).
Will take an eye on this.

Chapoly1305, 2023-12-07:
> Hmm, I've been requesting data for over 280 keys at once lately, once per hour, haven't seen a problem. But as you seem to be sure ... I wasn't actually watching the response I must admit (pushing the response right into db). Will take an eye on this.

Much appreciated for your feedback, it turns out to be a coding issue of my program. Seems there is no a noticeable limit for this query yet. I have just given another test,  **1552 keys in one request and received all of their reports in total of 1,576,864 Bytes**.