biemster/FindMy #34: better tutorial

ErfanDL, 2023-11-10:
Hello and thank you for this great project. I had a request, if possible, please post a more complete tutorial with all the steps and screenshots. I worked for several days on the correct implementation of this project, but I encounter many errors in different parts.

biemster, 2023-11-10:
Please open separate issues for the problems you encounter, the steps in the README work great on my machine!
Also, as this is just a proof of concept, a bit of experimentation is required. That's way more fun and you'll learn a lot, and increases the chances for me that people will contribute on the technical side of things.

ErfanDL, 2023-11-10:
> Please open separate issues for the problems you encounter, the steps in the README work great on my machine! Also, as this is just a proof of concept, a bit of experimentation is required. That's [...]

thanks for reply. I always get this error after running `anisette-v3-server/anisette-v3-server & cd pypush ; python3 examples/openhaystack.py ; cd ..; killall anisette-v3-server`

Traceback (most recent call last):
  File "/home/dietpi/pypush/examples/openhaystack.py", line 37, in <module>
    search_party_token = r['delegates']['com.apple.mobileme']['service-data']['tokens']['searchPartyToken']
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'service-data'
[1]   Done                    anisette-v3-server/anisette-v3-server  (wd: ~)
(wd now: ~/pypush)
[main(----) INF] Received signal 15. Shutting down.
Warning (thread: main): leaking eventcore driver because there are still active handles
  FD 8 (streamListen)
Use '-debug=EventCoreLeakTrace' to show where the instantiation happened
Warning (thread: main): leaking eventcore driver because there are still active handles
  FD 8 (streamListen)
Use '-debug=EventCoreLeakTrace' to show where the instantiation happened

biemster, 2023-11-10:
Is the anisette server running properly?

ErfanDL, 2023-11-10:
> Is the anisette server running properly?

I installed with this steps:

apt update && apt install --no-install-recommends -y ca-certificates ldc git clang dub libz-dev libssl-dev
git clone https://github.com/Dadoum/anisette-v3-server.git; cd anisette-v3-server
DC=ldc2 dub build -c "static" --build-mode allAtOnce -b release --compiler=ldc2
stat anisette-v3-server

biemster, 2023-11-10:
And if you run the `anisette-v3-server` executable, and do `$ curl localhost:6969`, do you get some proper result?

ErfanDL, 2023-11-10:
> And if you run the `anisette-v3-server` executable, and do `$ curl localhost:6969`, do you get some proper result?

yes I get some result.

ErfanDL, 2023-11-10:
> then `pypush` is still under heavy development, you'll probably have to focus your debugging there. Can you run only that? What errors do you get from there?

you mean `python3 examples/openhaystack.py` ?

biemster, 2023-11-10:
Yes that, it seems `pypush` is not able to login and retrieve the `search-party-token`

ErfanDL, 2023-11-10:
> Yes that, it seems `pypush` is not able to login and retrieve the `search-party-token`

its showing;

`dietpi@DietPi:~/pypush/examples$ python3 ./openhaystack.py
Traceback (most recent call last):
  File "/home/dietpi/pypush/examples/./openhaystack.py", line 5, in <module>
    import icloud.gsa as gsa
ModuleNotFoundError: No module named 'icloud'`

biemster, 2023-11-10:
could you start it from the root of the repo: `dietpi@DietPi:~/pypush/$ python3 examples/openhaystack.py`

biemster, 2023-11-10:
and probably make a `config` directory first: `dietpi@DietPi:~/pypush$ mkdir config`

ErfanDL, 2023-11-10:
Traceback (most recent call last):
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connection.py", line 203, in _new_conn
    sock = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/util/connection.py", line 85, in create_connection
    raise err
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/util/connection.py", line 73, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 791, in urlopen
    response = self._make_request(
               ^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 497, in _make_request
    conn.request(
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connection.py", line 395, in request
    self.endheaders()
  File "/usr/local/lib/python3.11/http/client.py", line 1277, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/local/lib/python3.11/http/client.py", line 1037, in _send_output
    self.send(msg)
  File "/usr/local/lib/python3.11/http/client.py", line 975, in send
    self.connect()
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connection.py", line 243, in connect
    self.sock = self._new_conn()
                ^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connection.py", line 218, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0x7f666422da50>: Failed to establish a new connection: [Errno 111] Connection refused

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/adapters.py", line 486, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 845, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/util/retry.py", line 515, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='localhost', port=6969): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f666422da50>: Failed to establish a new connection: [Errno 111] Connection refused'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dietpi/pypush/examples/openhaystack.py", line 35, in <module>
    r = icloud.login(USERNAME, PASSWORD, delegates=["com.apple.mobileme"])
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/pypush/icloud/__init__.py", line 36, in login
    g = gsa.authenticate(username, password)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/pypush/icloud/gsa.py", line 362, in authenticate
    r = authenticated_request(
        ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/pypush/icloud/gsa.py", line 190, in authenticated_request
    "cpd": _generate_cpd(),
           ^^^^^^^^^^^^^^^
  File "/home/dietpi/pypush/icloud/gsa.py", line 117, in _generate_cpd
    cpd.update(generate_anisette_headers())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/pypush/icloud/gsa.py", line 176, in generate_anisette_headers
    a = _generate_remote_anisette(ANISETTE)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/pypush/icloud/gsa.py", line 167, in _generate_remote_anisette
    h = json.loads(requests.get(url, timeout=5).text)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/adapters.py", line 519, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='localhost', port=6969): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f666422da50>: Failed to establish a new connection: [Errno 111] Connection refused'))

biemster, 2023-11-10:
hm darn, how does your `ANISETTE` link look in `icloud/gsa.py`? (`$ grep "ANISETTE =" icloud/gsa.py`)

ErfanDL, 2023-11-10:
server already running in another terminal

Traceback (most recent call last):
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 468, in _make_reque                           st
    self._validate_conn(conn)
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 1097, in _validate_                           conn
    conn.connect()
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connection.py", line 642, in connect
    sock_and_verified = _ssl_wrap_socket_and_match_hostname(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connection.py", line 783, in _ssl_wrap_socke                           t_and_match_hostname
    ssl_sock = ssl_wrap_socket(
               ^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/util/ssl_.py", line 471, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(sock, context, tls_in_tls, server_hostname)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/util/ssl_.py", line 515, in _ssl_wrap_socket                           _impl
    return ssl_context.wrap_socket(sock, server_hostname=server_hostname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/ssl.py", line 517, in wrap_socket
    return self.sslsocket_class._create(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/ssl.py", line 1075, in _create
    self.do_handshake()
  File "/usr/local/lib/python3.11/ssl.py", line 1346, in do_handshake
    self._sslobj.do_handshake()
TimeoutError: _ssl.c:975: The handshake operation timed out

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/adapters.py", line 486, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 845, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/util/retry.py", line 470, in increment
    raise reraise(type(error), error, _stacktrace)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/util/util.py", line 39, in reraise
    raise value
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 791, in urlopen
    response = self._make_request(
               ^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 492, in _make_reque                           st
    raise new_e
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 470, in _make_reque                           st
    self._raise_timeout(err=e, url=url, timeout_value=conn.timeout)
  File "/home/dietpi/.local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 371, in _raise_time                           out
    raise ReadTimeoutError(
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='gsa.apple.com', port=443): Read timed out. (re                           ad timeout=5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dietpi/pypush/./examples/openhaystack.py", line 35, in <module>
    r = icloud.login(USERNAME, PASSWORD, delegates=["com.apple.mobileme"])
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/pypush/icloud/__init__.py", line 36, in login
    g = gsa.authenticate(username, password)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/pypush/icloud/gsa.py", line 362, in authenticate
    r = authenticated_request(
        ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/pypush/icloud/gsa.py", line 203, in authenticated_request
    resp = requests.post(
           ^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/api.py", line 115, in post
    return request("post", url, data=data, json=json, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/.local/lib/python3.11/site-packages/requests/adapters.py", line 532, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='gsa.apple.com', port=443): Read timed out. (read t                           imeout=5)
dietpi@DietPi:~/pypush$ python3 ./examples/openhaystack.py
Username: **********@gmail.com
Password: *************
Traceback (most recent call last):
  File "/home/dietpi/pypush/./examples/openhaystack.py", line 37, in <module>
    search_party_token = r['delegates']['com.apple.mobileme']['service-data']['tokens']['searchPartyToken']
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'service-data'

ErfanDL, 2023-11-10:
After a few tests:

Traceback (most recent call last):
  File "/home/dietpi/pypush/./examples/openhaystack.py", line 37, in <module>
    search_party_token = r['delegates']['com.apple.mobileme']['service-data']['tokens']['searchPartyToken']
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'service-data'

olivluca, 2023-11-10:
Could it be that your anisette server isn't working properly?
I hate to add more confusion here, but could you try the docker version? That's what I'm using (didn't feel like compiling it myself) and it works.

biemster, 2023-11-10:
yeah what @olivluca said, could you copy the "some result" you mentioned here?

biemster, 2023-11-10:
Also just saw this now in your log:
```
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='gsa.apple.com', port=443): Read timed out.
```
does your pi have internet access or only local lan?

ErfanDL, 2023-11-10:
> Also just saw this now in your log:
> 
> ```
> requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='gsa.apple.com', port=443): Read timed out.
> ```
> 
> does your pi have internet access or only local lan?

Yes, it has an internet connection

ErfanDL, 2023-11-10:
> Could it be that your anisette server isn't working properly? I hate to add more confusion here, but could you try the docker version? That's what I'm using (didn't feel like compiling it myself) and it works.

thanks for reply. I will test and tell you the result

ErfanDL, 2023-11-10:
I installed the server on Docker and it still gives the same error

Traceback (most recent call last):
File "/home/dietpi/pypush/./examples/openhaystack.py", line 37, in
search_party_token = r['delegates']['com.apple.mobileme']['service-data']['tokens']['searchPartyToken']
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'service-data'

ErfanDL, 2023-11-10:
aha ! finally worked I asked it from GPT4 and he/she said:

It looks like you're encountering a KeyError in your Python code, specifically on the line:

python

`search_party_token = r['delegates']['com.apple.mobileme']['service-data']['tokens']['searchPartyToken']`

The error message indicates that the key 'service-data' is not present in the dictionary accessed by r['delegates']['com.apple.mobileme'].

To troubleshoot this issue, you should add some error checking to ensure that the keys you are trying to access actually exist in the dictionary. You can use the get method for dictionaries, which allows you to provide a default value if the key is not found. Here's an example of how you can modify the problematic line:

python

`search_party_token = r.get('delegates', {}).get('com.apple.mobileme', {}).get('service-data', {}).get('tokens', {}).get('searchPartyToken')`

This way, if any of the keys are missing, the corresponding get method will return None instead of raising a KeyError. You can then add additional checks or handle this situation based on your requirements.

biemster, 2023-11-10:
the problem is actually upstream from that, you're having connection issues with the Apple servers. That's why the key "service-data" is not present in your dictionary, there is probably no dictionary at all.

ErfanDL, 2023-11-11:
> the problem is actually upstream from that, you're having connection issues with the Apple servers. That's why the key "service-data" is not present in your dictionary, there is probably no dictionary at all.

but with `search_party_token = r.get('delegates', {}).get('com.apple.mobileme', {}).get('service-data', {}).get('tokens', {}).get('searchPartyToken')` the problem solved !

biemster, 2023-11-11:
Really? are you getting FindMy reports now or is the error now silently ignored?

ErfanDL, 2023-11-11:
> Really? are you getting FindMy reports now or is the error now silently ignored?
yeah its working and registering config in openhaystack.json

dietpi@DietPi:~/pypush$ python3 ./examples/openhaystack.py
Using saved config...
Error fetching locations (ratelimit?):  401 {'Server': 'AppleHttpServer/78689afb4479', 'Date': 'Sat, 11 Nov 2023 10:30:24 GMT', 'Content-Length': '0', 'Connection': 'keep-alive', 'X-Responding-Instance': 'acsnservice:2:prod-p00-acsnservice--partial-cf86dfff7-m9r66:80:2499B1540:92f6f6e4a3a6', 'WWW-Authenticate': 'X-MobileMe-AuthToken realm="Newcastle", Basic realm="Newcastle"', 'Strict-Transport-Security': 'max-age=31536000; includeSubDomains;', 'x-apple-user-partition': '0', 'via': 'xrail:st53p00ic-qujn13071302.me.com:8301:23R539:grp60,631194250daa17e24277dea86cf30319:42bb11b966e814d82960b7415e9f9025:defra2', 'X-Apple-Request-UUID': 'd570fc0e-1cb6-46b7-86e5-a40c63cd696a', 'access-control-expose-headers': 'X-Apple-Request-UUID,Via', 'X-Apple-Edge-Response-Time': '321'}

biemster, 2023-11-11:
That error is a bit undescriptive indeed, the Apple servers are not accepting your credentials. Very likely your `search-party-token`, do you have a proper one in your `config/openhaystack.json`? (don't copy it here, just mention if it's a really long string like `EAAMAAAABLwIAAAAAGVBSYQRDmdzLml.... etc`, or `None` or empty or something)

ErfanDL, 2023-11-12:
@biemster 
I installed a clean dietpi os and test all again.
after run the bellow code show `Logged in!` message

`dietpi@DietPi:~/pypush$ python3 ./examples/openhaystack.py`

but show this error:
`Error fetching locations (ratelimit?):  401`

in the openhaystack.json 
`"search_party_token": null,`

biemster, 2023-11-12:
Alright so definitely `pypush` is not giving you a search party token. That can have multiple reasons, and if I were you I would focus on getting that running properly first (so leave these FindMy scripts for now, run your anisette server and experiment with `python3 examples/openhaystack.py` until you have a `search_party_token` in your json. Don't forget to remove your `openhaystack.json` after every experiment otherwise it just reuses the data in there instead of generating a new one)

I'm not an expert on pypush, but they have an excellent discord server where people help out swiftly. You can find the invite on https://github.com/beeper/pypush (and please report back here when you solve your issue!)

ErfanDL, 2023-11-12:
> Alright so definitely `pypush` is not giving you a search party token. That can have multiple reasons, and if I were you I would focus on getting that running properly first (so leave these FindMy [...]

thanks for reply. and final question: do I need to enable SMS 2FA ?

biemster, 2023-11-12:
2FA is mandatory for new apple IDs if I'm not mistaken, so if this is your first apple device that has to go via SMS indeed.

olivluca, 2023-11-12:
Currently SMS 2FA isn't working in pypush, you should have another device (iphone, mac, virtual mac) already validated that you can use to validate the login with pypush.

ErfanDL, 2023-11-12:
> Currently SMS 2FA isn't working in pypush, you should have another device (iphone, mac, virtual mac) already validated that you can use to validate the login with pypush.

I installed macOS 10.15 on vmware but when I try login to iCloud keep asking me terms of condition repeatedly.

[image]

biemster, 2023-11-12:
Yes I had that issue as well, you'll have to connect your Apple ID with a phone number and activate 2FA. You can do that via their website, although it was a while ago for me so I'm not fully sure anymore if I did that through icloud or appleid.

ErfanDL, 2023-11-14:
> Yes I had that issue as well, you'll have to connect your Apple ID with a phone number and activate 2FA. You can do that via their website, although it was a while ago for me so I'm not fully sure anymore if I did that through icloud or appleid.

finally I got the token. now when I run `./request_reports.py` get error:

dietpi@DietPi:~/FindMy$ ./request_reports.py
Traceback (most recent call last):
  File "/home/dietpi/FindMy/./request_reports.py", line 69, in <module>
    res = json.loads(r.content.decode())['results']
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

ErfanDL, 2023-11-14:
> Can you add a debug line before line 69:
> 
> ```python
> print(r.content.decode())
> ```
> 
> and post the result?

Traceback (most recent call last):
  File "/home/dietpi/FindMy/./request_reports.py", line 70, in <module>
    res = json.loads(r.content.decode())['results']
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

biemster, 2023-11-14:
So it's empty? this is the same trace as in your previous post

ErfanDL, 2023-11-14:
> So it's empty? this is the same trace as in your previous post

yeah, it did not differ from the previous post

ErfanDL, 2023-11-14:
I think it's better to abandon this project :( or I have to save money to buy a Macbook!

biemster, 2023-11-14:
> > So it's empty? this is the same trace as in your previous post
> 
> yeah, it did not differ from the previous post

Then either you forgot to save, or the response from the request is empty. Could you add a bit of info text to that print statement?
Also, now you already have your macOS vm, you could use the `catalina` or `monterey` branch of this project. For that there is no anisette server or pypush necessary.

biemster, 2023-11-26:
The script changed quite a bit since your last try, and you should not need a Mac at all anymore. If you have a properly working `anisette-v3-server` or `pyprovision` install and an Apple ID with SMS 2FA activated, this should work now without problems.

ErfanDL, 2023-11-27:
> The script changed quite a bit since your last try, and you should not need a Mac at all anymore. If you have a properly working `anisette-v3-server` or `pyprovision` install and an Apple ID with SMS 2FA activated, this should work now without problems.

Hi thanks for new update. I downloaded the new version, but when I run pypush_gsa_icloud.py, it doesn't any respond

[image]

biemster, 2023-11-27:
Yeah why would you do that? Please just run `./request_reports.py`

ErfanDL, 2023-11-27:
> Yeah why would you do that? Please just run `./request_reports.py`

thanks for reply. now I got this error:
pyprovision is not installed, querying http://localhost:6969 for an anisette server
pyprovision is not installed, querying http://localhost:6969 for an anisette server
pyprovision is not installed, querying http://localhost:6969 for an anisette server
Traceback (most recent call last):
  File "/home/dietpi/FindMy/request_reports.py", line 72, in <module>
    auth=getAuth(regenerate=args.regen, second_factor='trusted_device' if args.trusteddevice else 'sms'),
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/FindMy/request_reports.py", line 34, in getAuth
    j = {'dsid': mobileme['dsid'], 'searchPartyToken': mobileme['delegates']['com.apple.mobileme']['service-data']['tokens']['searchPartyToken']}
                                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'service-data'

biemster, 2023-11-27:
did you add your phone number as a `trusted number` on appleid.apple.com?

ErfanDL, 2023-11-27:
> did you add your phone number as a `trusted number` on appleid.apple.com?

my appleid is old and I didn't enable 2FA on it.

biemster, 2023-11-27:
That's the issue you've been having all this time. This will never work without 2FA enabled, it's a restriction Apple put on. Please enable 2FA on your ID, and test if SMS verification works, and only then retry this or any OpenHaystack related project.
You could create a new ID if you prefer, one that you can discard later on easily, but also that will have to have SMS 2FA enabled.

ErfanDL, 2023-11-27:
> That's the issue you've been having all this time. This will never work without 2FA enabled, it's a restriction Apple put on. Please enable 2FA on your ID, and test if SMS verification works, and [...]

Well, Apple has boycotted our country and does not send SMS messages.

biemster, 2023-11-27:
I believe there are services where you can rent a phone number (I think even Mozilla has such a thing?) A quick search query "get temporary phone number for sms verification" gave me a whole bunch of results.

olivluca, 2023-11-27:
> I believe there are services where you can rent a phone number (I think even Mozilla has such a thing?) A quick search query "get temporary phone number for sms verification" gave me a whole bunch of results.

This is tricky: if such a number has been used many times, apple won't allow it. You'll have to find the freshest possible number. Don't ask me how I know that :wink:

ErfanDL, 2023-11-27:
> This is tricky: if such a number has been used many times, apple won't allow it. You'll have to find the freshest possible number. Don't ask me how I know that 😉

I can do it right now, but these numbers are disposable

olivluca, 2023-11-27:
So what? Once you're authenticated you don't need to receive the sms anymore (or at least not for a while).

ErfanDL, 2023-11-28:
> So what? Once you're authenticated you don't need to receive the sms anymore (or at least not for a while).

Well, if I try to log into the account again, the 2FA SMS will not come !

biemster, 2023-12-05:
I'm going to close this for now, as it is mostly an issue with Apple 2FA and geo restrictions which are beyond my control.

closed 2023-12-05 by biemster