biemster/FindMy #37: Crypto vs Cryptodome

olivluca, 2023-12-04:
I'm not really into python cryptography, but wanting to try the latest few commits, I saw an error importing SHA256. I couldn't find a debian package providing it, but I found `python3-pycryptodome`, so I made this change
```

diff --git a/pypush_gsa_icloud.py b/pypush_gsa_icloud.py
index d6d84e4..528b010 100644
--- a/pypush_gsa_icloud.py
+++ b/pypush_gsa_icloud.py
@@ -12,7 +12,7 @@ from datetime import datetime
 import srp._pysrp as srp
 from cryptography.hazmat.primitives import padding
 from cryptography.hazmat.primitives.ciphers import Cipher, algorithms, modes
-from Crypto.Hash import SHA256
+from Cryptodome.Hash import SHA256
 
 # Created here so that it is consistent
 USER_ID = uuid.uuid4()

```

Searching cryptodome I found [this bug](https://github.com/apprenticeharper/DeDRM_tools/issues/1957) so maybe it's a good idea to import from Cryptodome instead of from Crypto.

Systm21, 2023-12-04:
`pip install cryptography`

Should work.

olivluca, 2023-12-04:
Sure, but I prefer to use debian packages whenever possible.

Chapoly1305, 2023-12-05:
I vote for not a bug of this project and this can be closed. You shall use virtual environmental (venv) for individual projects. Using apt to install isn't a good idea anyway due to potential version conflicts and broken dependency.

biemster, 2023-12-05:
> I vote for not a bug of this project and this can be closed. You shall use virtual environmental (venv) for individual projects. Using apt to install isn't a good idea anyway due to potential version conflicts and broken dependency.

I tend to disagree here, I too prefer to manage my python packages using the os package manager.

olivluca, 2023-12-05:
I cannot ensure 100% that the change Crypto -> Cryptodome works, but I could authenticate (surprisingly it just asked me the username and password but no authentication code) and get the reports.

biemster, 2023-12-05:
> surprisingly it just asked me the username and password but no authentication code

That's because of the anisette.

olivluca, 2023-12-05:
But on subsequent calls it doesn't even ask for username/password.
I checked and while the previous openhaystack.json and the new auth.json have the same ds_prs_id/dsid, they have a different search party token.

biemster, 2023-12-05:
The first call is to get the search party token, and when the anisette data used for that is from a trusted device, it will not ask for 2FA. The subsequent calls use the cached search party token from the first call as password and the dsid as username.

olivluca, 2023-12-05:
Sorry for abusing thins bug, but where is the anisette data stored? In the anisette server (I'm using the dockerized one)?.

biemster, 2023-12-05:
The anisette server is faking an apple device, with the serials and stuff (ADI) stored in $HOME/.config/anisette-v3/adi.pb and device.json. Those numbers together generate a validation code like OTP, which is then going through a whole bunch of Mixed Boolean Arithmetic to generate two anisette headers for the request. One of the is a machine serial that is quite static over time, the other one is the otp which changes a lot.

olivluca, 2023-12-05:
Ah, thank you, I see those files inside the container, I thought they would be stored in the mounted volume (which is empty).

olivluca, 2023-12-05:
I'm trying (unsuccessfully but that will be another bug) to use provision and I see that if I install cryprodome with pip in a venv, the correct import is Crypto, while with the debian package is Cryptodome. For the time being, while I'm experimenting with provision but I want to use the, working, system python I did this:

```
diff --git a/pypush_gsa_icloud.py b/pypush_gsa_icloud.py
index d6d84e4..cd21b89 100644
--- a/pypush_gsa_icloud.py
+++ b/pypush_gsa_icloud.py
@@ -12,7 +12,10 @@ from datetime import datetime
 import srp._pysrp as srp
 from cryptography.hazmat.primitives import padding
 from cryptography.hazmat.primitives.ciphers import Cipher, algorithms, modes
-from Crypto.Hash import SHA256
+try:
+  from Crypto.Hash import SHA256
+except:
+  from Cryptodome.Hash import SHA256
 
 # Created here so that it is consistent
 USER_ID = uuid.uuid4()

```

biemster, 2023-12-05:
the original pypush code uses `pbkdf2.PBKDF2(p, salt, iterations, hashlib.sha256).read(32)`, then you just need to `import hashlib` and can leave out SHA256 from Crypto.Hash. For me that did not work for some reason, but maybe for you it will?

olivluca, 2023-12-05:
Nope, that doesn't work because hashlib.sha256 has no digest_size (only an instance created with it has the attribute).
I don't understand how the original code could possibly work.

biemster, 2023-12-05:
Yes that's my error too! :shrug:

maelp, 2025-01-12:
```
 File "/Users/primet/work/gouach/code/experiments/FindMy/pypush_gsa_icloud.py", line 16, in <module>
    from Crypto.Hash import SHA256
ModuleNotFoundError: No module named 'Crypto'
```

even after installing `cryptography` or `pycryptodome`, any idea?

biemster, 2025-01-12:
Can you try @olivluca's fix: `from Cryptodome.Hash import SHA256` ?

maelp, 2025-01-12:
okay, this works (requires installing `pycryptodomex`)

perhaps you could use `uv` in your repo to add packages and pinpoint them to the correct version?

using `uv script run` trick you can even add the dependencies in the scripts

biemster, 2025-01-12:
yeah I learned about `uv` a couple weeks ago as well, looks promising. However the code in this repo should be used as reference and not as a full production application, so this is not priority for me..

jcardus, 2025-09-04:
Installing cryptography didn't work for me.
I only got rid of the error with this:

```
pip uninstall crypto
pip uninstall pycryptodome
pip install pycryptodome
```