biemster/FindMy #39: --trusteddevice doesn't work

olivluca, 2023-12-05:
I'm trying to use provision and I think I have setup it correctly (if I get the anisette data from the container and keep the already obtained auth.json I can obtain the reports), but if I start with a clean slate (only the .so files in the anisette directory and no auth.json), I cannot get the 2FA with `--trusteddevice`

```
$ /home/luca/macintosh/venv/bin/python request_reports.py  --trusteddevice
Apple ID: xxxxxx@fivermail.com
Password: 
2FA required, requesting code
Traceback (most recent call last):
  File "/home/luca/macintosh/FindMy/request_reports.py", line 72, in <module>
    auth=getAuth(regenerate=args.regen, second_factor='trusted_device' if args.trusteddevice else 'sms'),
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/luca/macintosh/FindMy/request_reports.py", line 33, in getAuth
    mobileme = icloud_login_mobileme(second_factor=second_factor)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/luca/macintosh/FindMy/pypush_gsa_icloud.py", line 40, in icloud_login_mobileme
    g = gsa_authenticate(username, password, second_factor)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/luca/macintosh/FindMy/pypush_gsa_icloud.py", line 115, in gsa_authenticate
    trusted_second_factor(spd["adsid"], spd["GsIdmsToken"])
  File "/home/luca/macintosh/FindMy/pypush_gsa_icloud.py", line 253, in trusted_second_factor
    code = getpass.getpass("Enter 2FA code: ")
           ^^^^^^^^^^^^^^^
AttributeError: 'function' object has no attribute 'getpass'
```

If I omit the `--trusteddevice `I get a request for the 2FA but I cannot verify it works since I cannot receive the sms :wink: 

```
$ /home/luca/macintosh/venv/bin/python request_reports.py 
Apple ID: xxxxx@fivermail.com
Password: 
provisioning...
provisioning...
2FA required, requesting code
Enter 2FA code: 

```

biemster, 2023-12-05:
hm I did apparently not really test it. Could you remove one `getpass` from `getpass.getpass` and try again?

biemster, 2023-12-05:
Also, I see you did go with a venv? You're not in the os package manager camp anymore? :cry:

olivluca, 2023-12-05:
> hm I did apparently not really test it. Could you remove one `getpass` from `getpass.getpass` and try again?

That worked, thank you

> Also, I see you did go with a venv? You're not in the os package manager camp anymore? 😢

I could see no other way to use provsion :disappointed: 

```
~/macintosh/pyprovision$ python -m pip install .
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

```

biemster, 2023-12-05:
Thanks for testing, I'll push that fix shortly.

My pip has a `--break-system-packages`, doesn't yours?

olivluca, 2023-12-05:
Yes, the message says I could try it, but...but... :wink:

closed 2023-12-05 by olivluca