biemster/FindMy #40: Issue with IOS 17

shiprec, 2023-12-06:
Are you all seeing reporting issues from devices that have upgraded to IO17?  I haven't seen any updates come through since my phone upgraded but when I go back to 16 it works fine.

biemster, 2023-12-06:
I see you opened a similar issue in the openhaystack repo, but both of these projects have nothing to do with iOS. Can you elaborate exactly what works fine on 16?

biemster, 2023-12-06:
It's possible that apple ids connected to iOS17 require a different API to retrieve the reports. Do you have an error message of any kind from either openhaystack or the code in this repo?

redex557, 2023-12-06:
I can kind of confirm this issue, without any specific proof to confirm it so this might be biased. Last week I spend some time setting up new nrf51 devices and testing some alternative firmware with key rotation. Some devices never showed up and one took multiple hours to get a single report. 
All this while (at least) two iPhones where close, both running the latest iOS 17 update.

biemster, 2023-12-06:
That's very worrisome, maybe iOS17 can distinguish between real and cloned airtags?

shiprec, 2023-12-06:
Yeh there are no error messages, we just aren't getting any reports from IOS 17 reports...

biemster, 2023-12-14:
I'd like to keep this open if you don't mind, if future iOS and macOS devices know how to identify fake airtags this needs to be fixed if at all possible.

biemster, 2023-12-18:
Also the pypush discord is reporting this issue

Itheras, 2023-12-19:
Could this have something to do with this ? (https://github.com/seemoo-lab/openhaystack/issues/49#issuecomment-825186095)

biemster, 2023-12-19:
That's a great find @Itheras ! I was not aware of either of those, but they seem easy to implement. Having a micropython implementation is still high on my list, maybe it's time to finish that so I can easily experiment with these suggestions.

Systm21, 2023-12-23:
@Itheras @biemster  Are there any updates on this? it is noticeable that far fewer reports are coming in than before.

biemster, 2023-12-24:
No updates yet on this, also I lack an iOS17 device to do any testing..

loganmimaroglu, 2023-12-24:
I have been using OpenHaystack's HCI code for a RPi4B and have 2 iPhone 14 Pro Max's running iOS 17.1.2 and iOS 17.2.1 that I've placed near the device. I was getting, on average, 12 reports per 24 hours. I honestly thought this was quite bad but it is > 0.

The MR that @Itheras links to (https://github.com/seemoo-lab/openhaystack/pull/225#issue-2023652959) improved this quite a bit. The changes to the HCI code lead to me getting around 72 reports per 24 hours.

Itheras, 2023-12-24:
Nothing on my end i dont have a computer with me until after the 25th but if someone can check i was reading apples documentation it says we should be setting the Hint byte to the value of the 5th byte of the mac address I believe we always setting the Hint byte to 0. Can someone check i am on mobile if not I'll check after Christmas 😅.

loganmimaroglu, 2023-12-24:
> Nothing on my end i dont have a computer with me until after the 25th but if someone can check i was reading apples documentation it says we should be setting the Hint byte to the value of the 5th [...]

For the Lenze st17h66 source this appears to be true.

[image]

The HCI code I'm running also does this:
```
@staticmethod
def advertisement_template():
    adv = ""
    adv += "1e"  # length (30)
    adv += "ff"  # manufacturer specific data
    adv += "4c00"  # company ID (Apple)
    adv += "1219"  # offline finding type and length
    adv += "00"  # state
    for _ in range(22):  # key[6:28]
        adv += "00"
    adv += "00"  # first two bits of key[0]
    adv += "00"  # hint
    return bytearray.fromhex(adv)
```

Itheras, 2023-12-24:
It appears there are more differences than we thought, and any one of these could be causing the issue. I’ve attached the expected content. @loganmimaroglu  Thank you. Also great that you are seeing a positive result we should confirm if that is the actual fix.

[image]

Itheras, 2023-12-24:
Well there is 3 things to check the AdvertisementType, the status byte "the first 5bits have values that are expected" and the Hint that should not be 0.  It seems changing advertising type had good results but we still need to confirm and the other 2 needs testing. I will test after the 25th i am on mobile now 😅. Merry Christmas everyone.

Systm21, 2023-12-25:
That sounds like a solution, hopefully it won't affect the undercover mode of our tags. If it does, you can still keep the project alive. It would be a shame if it was already over before it had really started (where the first macless solutions are available).

Itheras, 2023-12-26:
I think I can call this confirmed the problem is the advertisement type.  would like if someone else can also confirm but we already have the same good results from @loganmimaroglu

Systm21, 2023-12-26:
What exactly is this hint byte doing that seems to be the problem?

Itheras, 2023-12-27:
@Systm21  I am not sure but it may be that is not the issue after all.  I noticed a difference changing the advertisement type from unconnectable scannable undirected to connectable scannable undirected  I have not tested the hint or status byte much still. we should look at all 3 factors to future proof I am just a little busy at the moment.

biemster, 2023-12-27:
Isn't the increase @loganmimaroglu sees coming from the broadcast freq change to 5Hz?

igrowstuff, 2023-12-27:
I have tried the linked HCI code using my RPI4 and I get no reports on my two iPhone 14 devices running 17.2.1. Only time I ever get any reports is if I turn the Bluetooth on, on my Mac.

loganmimaroglu, 2023-12-27:
> I have tried the linked HCI code using my RPI4 and I get no reports on my two iPhone 14 devices running 17.2.1. Only time I ever get any reports is if I turn the Bluetooth on, on my Mac.

I didn't even think about my Mac, oof. Let me test again...

Itheras, 2023-12-27:
Yeah can confirm changing advtype has no effect i saw more reports because neibor had family visiting.

Now trying changing status and hint.

Systm21, 2023-12-27:
I think it has something to do with our stealth mode. Apple is quite behind on the anti-stalking thing and I don't think they're going to completely change the basic system now.

igrowstuff, 2023-12-27:
> I think it has something to do with our stealth mode. Apple is quite behind on the anti-stalking thing and I don't think they're going to completely change the basic system now.

Maybe so. However I don't really understand why you would only enforce that on iOS 17. Other than the tag not being registered to an apple account the tags aren't stealth are they? Don't they just emulate a device in its separated state? The underlying system must be the same as it still works on older iOS and the latest macOS

Itheras, 2023-12-28:
@igrowstuff  Apple doesn't know if a tag is registered to an account or not. In function the lost mode advertising key tells apple nothing by design . It just gets uploaded. They must be doin a extra check with the data received that doesn't match what is expected. Think about this way why would your phone not upload a rotating separated advertisement key how would it even know who it belongs. When you want to know to who a tag belongs you need to have it physically with you and then the tag itself will give more info and this cannot be done from afar. So there is something we are missing with the data sent itself. Also airtags also work on single key mode without rotation if the battery is removed and reinstalled. Since it looses track of what key it should broadcasting it will default to single key.
[image]

igrowstuff, 2023-12-28:
I have tried setting the hint to key[5] and modifying the status byte and can't seem to generate any other reports.

humpataa, 2024-01-03:
Add me to the unlucky people. I have a few hundred advertising keys which my script used to requests reports for every hour. I am not sure when I updated my iPhone to iOS17, but I was always wondering why it didn't seem to be adding to the FindMy service for my 15+ devices. I am quite sure it was better early December and reports got fewer over the weeks. They have actually come to zero a couple of days ago, although my devices are still around and working. I will experiment a little more, but ... thinking about all of it I only see one possible reason:
Apple cannot change advertising because millions of devices would need an update. So they can only change the behaviour on their end, the reports getting in and out. I have read that AirTags change their keys once a day. Does anyone know if these keys are fixed per tag? I cannot imagine they are generated from scratch. If they are fixed, the keys (literally any number) of an AirTag may be send to the server at time when someone makes a tag his own (adds it to phone): the iPhone just asks the device for the list of keys it will be using and sends this list to the server making them "valid". Whenever some Apple device sends location reports they now know which keys belong to real AirTags and which not, save the data or not. This "making keys valid" procedure can happen for older already "bound" devices too, the iPhone may have always known all the keys of an AirTag, just hasn't registered them yet, but does so now because of an updated iOS system. Since more and more phones will have the latest software, more and more keys will come invalid.
I have read that Apple made a promise so that there is no way to know what AirTag belongs to whom, but I doubt that a) anyone believed that and b) does any user of AirTags actually care?
I can also imagine the procedure has not only changed because more and more fake devices kept spawning up but it may also because of Google's FindMy plans and the cooperation of both companies on that matter. Security and protection against stalking is a big issue and everything would be easier when they know who is who.
So if I am right and I am sure nobody will be able to hack into this "making keys valid" procedure I am certain that it is the sudden end of fake AirTags ...

humpataa, 2024-01-03:
No updated protocol document, date of last change is May 18, 2022.

humpataa, 2024-01-03:
@biemster after thinking about all this again, I believe you are right: nothing has changed really, but iOS devices now obviously check something that hasn't been taken seriously before. I am actually using the full Status byte and will try to stick to the document instead (using only bit 6-7 and keeping the others untouched). Same with the Hint: what does "Byte 5 of the Bluetooth address" mean? The MAC address? It is always set to 0 in my code ...

peterbaugh, 2024-01-03:
> @humpataa To comment on your edited message I doubt that ios17 has an extra check on the validity of the broadcasts it sees in terms of registered keys, since such a check is much easier implemented server side and would immediately work for all versions of ios and macos. The fact that only ios17 iphones (or also ipads?) are able to discern them leads me to believe there is something we are missing in the details of the BLE broadcast. I really hope that they are not also listening for the UWB stuff a real AirTag sends, because that will be very difficult to emulate.

I was considering the UWB feature, does anyone know if any of the third party made for iPhone type Airtags that do not have UWB still work ok?

humpataa, 2024-01-03:
Could someone point me to some document how AirTags make use of UWB? I have my doubts because I cannot find anything about this in FindMy network specification mentioned above.

humpataa, 2024-01-03:
> ... using the full Status byte and will try to stick to the document instead (using only bit 6-7 and keeping the others untouched). Same with the Hint: what does "Byte 5 of the Bluetooth address" mean? The MAC address? It is always set to 0 in my code ...

To quote myself: I have checked what my real AirTag advertises (separated) and set that as fixed Byte 2 (0x08) and set Byte 26 (Hint) to MAC addresses last byte. The device is advertising for half an hour and it indeed showed up in reports again! Unfortunately, I cannot make sure it is my own iPhone (iOS17) "catching" it, since it is quite a close neighborhood here ...

Systm21, 2024-01-03:
To check the UWB stuff:
You can clone bluetooth beacons with "nrf connect" on android for example. does anyone have an airtag or official licensed product, that can be cloned and then remove the batterys? If the reports come in without a problem the UWB stuff doesn't matter. 

I haven't had a close look at nrf connect yet, but it may be that it is better to play with the individual bytes than to permanently flash/reflash new tags.

malmeloo, 2024-01-03:
> To check the UWB stuff: You can clone bluetooth beacons with "nrf connect" on android for example. does anyone have an airtag or official licensed product, that can be cloned and then remove the [...]

Android will randomize the BLE address when advertising, and since the public key is encoded in that address it will be different from the cloned airtag. That said, I think it should still be possible to fetch those reports if you know what address it is using, as you only need the hash of that public key; you just won't be able to decrypt them. Have you already tested this by any chance?

Systm21, 2024-01-03:
I'm not into it that much, it was just an idea.

malmeloo, 2024-01-03:
Ah gotcha, thought you'd tried it before. It's some interesting food for thought though, I would kill to have my phone act as a beacon...

humpataa, 2024-01-03:
> To check the UWB stuff: You can clone bluetooth beacons with "nrf connect" on android for example. does anyone have an airtag or official licensed product, that can be cloned and then remove the batterys? If the reports come in without a problem the UWB stuff doesn't matter.

Interesting idea! I have both Android and iPhone with NRF Connect installed. And an original AirTag, an official Clone (Maginon Smart Tag) and of course my fake tags. Maybe I'll find some time tomorrow.

Apart from that, I have checked recent update details of iOS and it seems they have indeed made some changes to FindMy, claiming security fixes of course. So I have updated my iPhone to the latest it offered (17.2.1), refreshed two of my devices: one with my older code, one as close to the specification as possible (bits and hint). Both show up in the reports, but again, I cannot make sure it is my own phone doing this ...

> I haven't had a close look at nrf connect yet, but it may be that it is better to play with the individual bytes than to permanently flash/reflash new tags.

I am programming my devices using Espruino Javascript, so I don't have to flash all the time, makes it easy to try things.

humpataa, 2024-01-03:
Tried cloning an AirTag using Android NRF Connect: it is not being recognised, at least FindMy app on iPhone cannot see it anymore (neither nearby nor separated). The advertising does not look good anyway, it uses a public key (cannot change) which surely doesn't match specification. I will try disabling bluetooth on my phone and see if there is someone around reporting my devices or it is depending on mine overnight.

Systm21, 2024-01-03:
You told us, that you have a possible working Tag with ios17. How long does it take to retrieve the first reports? Maybe you can switch off your phone for a longer period to validate it... (Dont know if switching off bluetooth is enough)

humpataa, 2024-01-04:
Must have been someone in the neighborhood. I have made several tests today, setting type (scannable / connectable), bits of status byte, correct hint byte, even disabled name and set advertising to 2000ms – just like a real AirTag. Having my phone (iOS17), a MacBook and an iMac around does not result in reports. As soon as I leave the office, drive a few miles, reports come up – my tags are recognised. So I guess with more and more iPhones updating reports will get fewer and fewer making fake tags useless. I'm out of ideas, still doubt it is UWB however.

Itheras, 2024-01-04:
@humpataa yeah is definitely not related to UWB since I am testing with some find my compatible tags that don't use it and they work normally. I am starting to look into CSA mode.

Systm21, 2024-01-05:
Is it maybe possible, that our keys can be filtered out somehow and then be locked out from the system?

humpataa, 2024-01-05:
> Is it maybe possible, that our keys can be filtered out somehow and then be locked out from the system?

That was my first idea: newer iPhones send all the keys of their bound AirTags to the server. Even from older tags. That way, Apple's server now knows what reports to reject, because our tags never get registered. Not necessarily with knowing / saving to whom an AirTag belongs (but who knows and if so, will someone care?).
Older phones just keep sending without checking the validity first, resulting still in reports now and then. So our keys are not blocked, just not validated.
Since they are working [with Google](https://www.techradar.com/phones/android/android-find-my-device) to create a joint FindMy network, this might be a reason for changes.
However, technically I think this would only be possible if AirTags have a fixed number of keys that could be read and transferred to the server. Afair the specification does not clearly state if devices must be able to generate keys or simply store them.
Or – I doubt it, but still hope – it is just Apple's fault, they somehow screwed it up when adding latest AirTag features like sharing etc.

biemster, 2024-01-05:
I think we already discussed this, but the only validated keys hypothesis should be fairly straightforward to test by having a fake tag sending out a working airtag's broadcast right? I know on android we cannot set the bluetooth mac anymore, but the ESP32, st17h66 or the linux hci script should be able to send out the exact public key.

We might not even need to own an airtag, but just catch one in the wild and wait for the key to roll and start sending that broadcast again. (if that works it will confuse the actual owner of the airtag though, so let's not go overboard with that)

I'm still quite convinced that any key blocking or validating would be done server side, and not through a software update on specific devices.

Itheras, 2024-01-05:
@biemster I will test that today. but I am not sure how that would work how can they validate when most of the phones out there are old or in older software versions. newer phones won't have nothing to validate against since those older phones and firmwares don't upload the necessary information do even do a check . so anyone under ios17 would have findmy experience severely impacted as time goes on.

biemster, 2024-01-05:
@Itheras great! I'm not sure if I understand the rest of your comment though, I tried to explain my point of view that I find it unlikely that the keys are checked for validity now. That's your opinion too right?
I highly suspect that we are missing a bob or a bit in our advertisement.

Systm21, 2024-01-05:
Maybe @adamcatley can help us with our problem, themes like he is the right contact person for our problem. 

If anyone can help us, it's him. 

[https://adamcatley.com/AirTag.html](url)

malmeloo, 2024-01-05:
Do AirTags also expose the Find My service UUIDs while in separated state? I have a protocol spec from 2020 which says that the they must be exposed while in pairing state, but I don't think it says anything about it for the other states. It could be that those UUIDs are always exposed - how else do you get a lost airtag to make a sound? Maybe iOS 17 just adds an additional check to see whether those UUIDs are visible before uploading the report.

I've written a utility script [here](https://github.com/malmeloo/FindMy.py/blob/main/examples/device_scanner.py) that scans for nearby devices and shows what are essentially all identifying data points, at least to my knowledge. I don't own any real AirTags myself, but it should include the UUIDs in its output. Maybe someone with a real tag can test it :)

humpataa, 2024-01-05:
> Do AirTags also expose the Find My service UUIDs while in separated state? I have a protocol spec from 2020 which says that the they must be exposed while in pairing state, but I don't think it says [...]

Interesting question. NRF Connect says a separated AirTag offers this:

Generic Access UUID: 0x1800
Generic Attribute UUID: 0x1801
UUID: 7dfc8000-7d1c-4951-86aa-8d9728f8d66c
UUID: 7dfc7000-7d1c-4951-86aa-8d9728f8d66c
UUID: 7dfc6000-7d1c-4951-86aa-8d9728f8d66c
UUID: 7dfc9000-7d1c-4951-86aa-8d9728f8d66c

The 4 longer seem to be Apple specific services, may indeed be used to let the AirTag play a sound ([see here](https://www.metabaseq.com/apple-airtags-unawanted-tracking-capabilities/)). Wether they are essential to iOS17 for sending location packets needs to be tested.

Systm21, 2024-01-05:
Apple Tags are a thing to have an Eye on, but i think the Cheap China "Works with Apple Find My" Tags are more interesting at this moment. The cheap ones do not have the UWB stuff and are better to research, i think.

Systm21, 2024-01-05:
> Interesting question. NRF Connect says a separated AirTag offers this:
>
> Generic Access UUID: 0x1800
> Generic Attribute UUID: 0x1801
> UUID: 7dfc8000-7d1c-4951-86aa-8d9728f8d66c
> UUID: [...]

Interesting point, what if someone adds cloned/randomized UUIDs to a fake tags, from the documentation the 9000 uuid should be the findmy identfier

malmeloo, 2024-01-06:
> Interesting question. NRF Connect says a separated AirTag offers this:
>
> Generic Access UUID: 0x1800 Generic Attribute UUID: 0x1801 UUID: 7dfc8000-7d1c-4951-86aa-8d9728f8d66c UUID: [...]

Interesting! The older protocol spec I found lists different UUIDs, but it does mention that they will change in an updated document. Is anyone willing to share a copy of the 2022 version?

Anyway, I'm hoping that simply exposing those UUIDs should be enough; if the rest of my doc is up-to-date then only one of those services should be accessible to non-owners of the device, and that one only allows for sound control which is an optional feature. Also, it's one thing to just receive broadcasts, but another to actually communicate with the device which would be far less reliable; therefore I think (hope) it's just a matter of tweaking the "public image" of the tag to match the official ones.

I'll see if I can get my hands on a 3rd party tag and have someone pair it for me in the coming days; I think the easiest way to go forward right now is to just clone the BLE address/payload of a real tag and work from there.

Systm21, 2024-01-06:
I have read a lot the last hours, just about uuids and what they do... It seems, that they get randomly renewed by the Tag itself (not the identifier part). The good part in this is: The Server doesn't know the random payload of the uuid, so we can fill it with something valid, that comes from an active Airtag (or completely random maybe). 

What I also thought about: in ios17 it was introduced that you can now share airtags between iphones. Maybe they need the UUID for this and the blocking of our tags was just a coincidence. 

I mean, what interest would Apple have in willfully blocking us? Openhaystack was already big in the press in 2021 and nothing happened. Then there are the extremely cheap itags from the Far East. The licenses can't be that expensive if such a part is already available for ~€4.  I'm not deliberately blocking a few tinkerers who are having fun with their findings. I mean, how many people should copy this? 

By the way, there are also some people in the apple forum who can no longer find their original airtags after the update to ios17.

humpataa, 2024-01-06:
I have tried adding UUID 7dfc9000-.... to the advertising. NRF Connect shows "Apple reserved service", so it is visible. Key advertising and hint byte looking good as well. Not getting noticeably better. 

I am sure most (all?) chinese "clones" are not using Apple's network. I have some on my desk looking exactly like AirTags, but are just plain BLE devices, completely useless when they get out of range of the phone.

> By the way, there are also some people in the apple forum who can no longer find their original airtags after the update to ios17.

That's what I thought: they may have really screwed it up. It still looks good for many, because old iOS are still around.

Systm21, 2024-01-06:
> I am sure most (all?) chinese "clones" are not using Apple's network. I have some on my desk looking exactly like AirTags, but are just plain BLE devices, completely useless when they get out of range of the phone.
> 
> >

U told us, that you have an Maginon Tag, this Tag also works with the network and i can give you 100% they all use the findmy network from apple. The Itags must be added in a bit different way, than the fancy airtag popup, but it works also... 
Not every from this cheap tags supports it, thats right. It must be advertised with "Works with Apple Find My". 

please don't put so much hope in NRF Connect, I don't know if the results are really usable. It was just an idea and is completely unvalidated. It's possible that you won't be able to recreate airtags with it.

humpataa, 2024-01-06:
Yeah, I use NRF Connect only to check devices. Cloning does not work properly (at least not enough to clone AirTags), so I have checked by altering the code I use with my tags The Maginon I mentioned is an official clone, but many of the chinese are not. You can check [here](https://mfi.apple.com/account/accessory-search).

Systm21, 2024-01-06:
The Chinese ones are official too, so i wrote, that the licences cant be that expensive. 
They are cool, but not working with Android, as the original Airtags. 

I've also seen, that Lenze is offering an [FindMy SDK](https://lenzetech.com/findmy) for their Chips, so it should not very difficult for the dealers.

Itheras, 2024-01-06:
ok i cloned a Chinese official tag. Something really interesting happened. The cloned tag was visible to the find my app and marked as just seen as the original tag . The status bit was set to 0x20 the app sees this as full battery but reading all the documentation i thought it was supposed to be 0x10. Now the interesting part i left home and left the tag on top of my sonoma mac mini and i get no new reports. So find my updates only while in range. Now i will test with the original when i get back later today.

[82 more replies]