biemster/FindMy #42: Approach to 2 FA on Linux + Python Errors

liebrandapps, 2023-12-11:
Is it possible to fix the int.from_bytes() calls? (4 times - see below in request_reports.py)
    confidence = int.from_bytes(data[8:9], 'big')
    status = int.from_bytes(data[9:10], 'big')
    priv = int.from_bytes(base64.b64decode(privkeys[report['id']]), 'big')
    timestamp = int.from_bytes(data[0:4], 'big') +978307200
Please also fix the getpass.getpass() call (remove the leading getpass.)

Thank you!

Running this on linux, I am not getting the 2FA prompt unless I add --trusteddevice as parameter. I read elsewhere that this is a problem on linux... so it now works fine for me...

biemster, 2023-12-11:
I doubt there is anything wrong with them, I'm running this on linux too.
You mention you don't get a 2FA prompt, do you get an error message?

liebrandapps, 2023-12-11:
Nothing really wrong. If I run request_reports.py w/o any parameters, I am prompted for the 2nd factor, but never receive anything on my devices. If I use the parameter "--trusteddevices", the 2nd factor seems to be submitted via a popup on the apple devices (instead of SMS). That works just fine. ...

biemster, 2023-12-11:
Do you have SMS 2FA enabled?

liebrandapps, 2023-12-11:
I have 2 FA enabled, not sure that I have a choice between SMS and trudest device ...

biemster, 2023-12-11:
currently pypush is not working properly due to Apple blocking Beeper Mini, and since the auth code is taken from there it's possible that SMS 2FA doesn't work at the moment. I'll have to test this though

biemster, 2023-12-11:
werkt 't nou?

liebrandapps, 2023-12-12:
Yes, looks like it is working now (Python 3.10.2). Thank you!

ngxson, 2023-12-30:
Just a suggestion, you may not even need 2FA, just create a burner account (don't use your main icloud account)

In fact, burner account is actually recommended if you want to sideload apps (via altstore for example), so if apple decide to block the account, you lose nothing.

liebrandapps, 2023-12-30:
Agreed ... To my knowledge, the only way to get a non 2FA id is via iTunes on Windows. A few days ago, this was "just" hanging, no idea. I created a separate ID on the Apple web site. This has 2FA, but refuses to work ("Account limit reached"). My normal Apple ID works fine. It asked for the user id / 2FA initially (like 3 weeks ago), since then it uses the creds in auth.json w/o prompting.

biemster, 2023-12-30:
I seem to remember I needed to add 2FA to the apple id I created just for this, otherwise it could not log in to iCloud. Which is a requirement to get the `search-party-token`.
However I've heard good things about Apple Music accounts created either on the web or on android, it's somewhere on my list to check if they are readily usable for this as well.

closed 2024-09-30 by biemster