biemster/FindMy #43: Add a license

malmeloo, 2023-12-21:
Hi!

I'd like to make some contributions in order to make it easier to adopt the features of this tool into other projects. Before doing that though, could you specify a license for this project? That way, people are aware of the conditions under which your project can be used.

biemster, 2023-12-21:
Thanks for your interest! I don't accept any non-bugfix PRs though, and the conditions are do whatever you want but a mention would be nice. Also, my name is like the polder that's just how we say it.

closed 2023-12-21 by biemster

malmeloo, 2023-12-21:
Thank you for the clarification! Don't worry about the mention, I'll make sure to include it.

biemster, 2024-02-12:
"Een beugel in de vestingbar" could be the sole condition in the license though, haven't been there way too long (sorry just noticed your bio, it was called INF in my days)

malmeloo, 2024-02-12:
I'm not sure if that's what people mean when they say "free as in beer" 😛