biemster/FindMy #44: Add docker deployment + API + maybe web UI

ngxson, 2023-12-30:
Hello, thank you for your work.

I made a fork of your repo here: https://github.com/ngxson/FindMy_Docker

On the `main` branch, I added docker-compose and Dockerfile so that the project can be run quickly.

On the `server` branch, I added:
- flask API server and some endpoints
- a simple web UI

I have no idea if I should do a pull request or not, because it's a big breaking change.

Feel free to discuss in the issue, ideally I want to merge all of my code to your repo.

What need to be decided:
- Should we keep the current sqlite3 database? My idea is that we delegate the storage to a 3rd party system, for example home assistant or traccar. This way, this repo is more like a "gateway" to retrieve data from icloud server.
- My code have a `setup.py` file used solely for generating `auth.json`, maybe we can find a way to improve this.
- Maybe we can install `pyprovision` in the docker build stage, so that we no longer need anisette server.

Again, thanks for your work!

biemster, 2023-12-30:
That's a great fork @ngxson , running it easily in a docker with a web ui is very useful.
I don't accept PRs for this project anymore, except bugfixes, because I consider this finished. Feel free to reuse the code for any project you'd like.
Thanks for the pingback, I think I will keep a list in the readme of interesting projects that are based on this.

adamast0r, 2024-01-31:
@ngxson it seems we can't open issues in your project, given that this project aim is to allow basic functionality would you allow the creation of issues? 

The web interface looks great, couple of minor improvements:

- web app: it would be great to be able to select which tag we want to show, and the time (last X hours) when the tags were identified?

- API: To improve any integrations (home assistant, node red, etc) with your service , you could send the results ordered by timestamp, where the first result would be the most recent? a parameter to only return the last known location of each tag would be amazing.

ngxson, 2024-01-31:
@adamast0r Thanks for noticing me, I didn't know that the issue was disabled (it was disabled by default)

Your idea is good, but recently the bug with iOS 17 does break anything, so for now I don't have much info to test. Will have a look when the iOS 17 issue get some updates. Please refer to https://github.com/biemster/FindMy/issues/40

closed 2024-09-30 by biemster