biemster/FindMy #45: FindMy webserver

rkreutz, 2024-01-06:
Hey there,

First of all, thanks for all the hard work you've been putting on this, much appreciated. 

I was wondering if there is a readily available webserver which we could run on a docker container or something on a Linux server which could serve the reports from a REST API or similar?

Thanks

biemster, 2024-01-07:
There are a couple forks and projects based on this work, I do vaguely remember one with the description you gave as well. I'm planning of keeping a list of interesting projects that spun out of this, but I did not get to that yet. If you search through the closed issues and PRs you will likely find want you need though.

rkreutz, 2024-01-07:
Went through closed issues and PRs but didn't find anything. Maybe it was in open haystack repo, will have a look there. Thanks

rkreutz, 2024-01-07:
After going through the forks I found 2 that have implemented a webserver to serve the reports, though both store the tags private keys on the webserver which is not ideal

https://github.com/ngxson/FindMy_Docker

https://github.com/Chapoly1305/FindMy

closed 2024-08-15 by rkreutz