biemster/FindMy #46: (urgent as reports will disappear): Can anybody help me download reports from a device listed in Find My app?

mackaaij, 2024-01-14:
Hi there,

I'm trying to get all location reports from a stolen iPhone which is listed in the "Find My" app.

Openhaystack doesn't work as it requires a plugin in Apple Mail - Apple Mail 16 doesn't support adding such plugins anymore...

The tool in this repository seems to generate ID's to be used for an arbitrary device. Can somebody help me with getting identifiers of my devices and change the code to use these instead of those generated by "generate_keys.py"?

The phone was stolen on 07-01-2024 (pickpocket in a train in the Netherlands) and as I understand Apple keeps reports for 7 days so the reports will already be purged.

Systm21, 2024-01-14:
You won't get any further with the Openhaystack, as the program is not designed for this and you always need the private key to decrypt the report. 
You should contact Apple very quickly, assuming they can't locate it for you, so they can at least block the serial number. Then you won't be able to do much with it.

mackaaij, 2024-01-14:
Thanks for the quick reply! In the research paper [Who Can Find My Devices?](https://petsymposium.org/popets/2021/popets-2021-0045.pdf) I found: "Using the respective private advertisement keys di, the owner device can then decrypt the received location reports."

As I understand the paper, you can can retrieve encrypted reports for any device with any Apple account for up to seven days. But only a device owner can decrypt the reports. This history is interesting for police investigation and might lead to recorded camera footage. A fresh trail is interesting as the police is willing to intercept as long as you have a specific location. I think it would be very interesting for people to quickly report a device missing and receive an alert as soon as the reports start coming in.

If the current program is too far off, hopefully the above is inspiring a new one :)

closed 2024-09-30 by biemster