biemster/FindMy #47: Amend/change look-back time for report on map POST /get_locations

Furtivo360, 2024-01-28:
Hi!

Thanks you so much for this project!

I've got a question: 

how and where can I amend the look-back time for the report?

Based on your documentation:

__POST /get_locations
This endpoint fetch the data from icloud server, decode it and return as an array.
Note:
hours is optional, default to 24 (meaning get all reports in the last 24 hours)__

thanks

biemster, 2024-01-29:
The actual look-back time is not functioning properly on Apple's side, either only the last 20 reports are returned or all reports available on the server (which is about 10 days worth)
The script can filter the retrieved reports to display only those from the last X hours, which is controlled by the `-H` parameter:
https://github.com/biemster/FindMy/blob/113ebf4017729b92a381624c1932065588c3ebde/request_reports.py#L41

Furtivo360, 2024-01-29:
Hi biemster! Thanks for the reply!

Can you explain me a bit better how I can amend that parameter myself? 

Thanks

biemster, 2024-01-29:
I'm not exactly sure what you mean, you can run something like `./request_reports.py -H 96` for 4 days worth of reports?

Furtivo360, 2024-01-29:
I mean, can I amend that parameter somewhere on the backend of the code in order to see only the report of last hour for example?

biemster, 2024-01-29:
There is just `request_reports.py`, no backend. You could set `-H 1` to *see* only the reports of the last hour, but Apple's servers only allow us to request either the last 20 reports, or everything it has.

Furtivo360, 2024-01-29:
Oh! Right! Thanks man 💪🏼

Furtivo360, 2024-01-29:
Hi again...sorry man but where can I find the " request_reports.py"? 
Thank you very much for your help

biemster, 2024-01-29:
What have you been running up to now? That's basically the only thing there!

Furtivo360, 2024-01-29:
I am using the docker image from https://github.com/ngxson/FindMy_Docker

All works fine but I don't understand (I don't find) where to change the -H  parameter

biemster, 2024-01-29:
Ah that explains a lot! That's not mine, please reach out there I don't know what they are doing..

closed 2024-09-30 by biemster