biemster/FindMy #50: Add requeremets.txt

fuad00, 2024-02-09:
just it

biemster, 2024-02-09:
ʙuʙaxş ne

closed 2024-02-09 by biemster

biemster, 2024-02-09:
Sorry that might have been a bit short answer, I can elaborate a bit.
This project is completed, and works as intended; a proof of concept to generate and retrieve reports from the FindMy network. It is fairly straightforward to get up and running, but does require a certain level of expertise.
I only accept bugfixes (or research like the iOS 17 issue as in #40), but for the rest, this is what it is, I don't have time for more tinkering.

maelp, 2025-01-12:
If needed you can use `uv` using this kind of tricks to make it easier https://simonwillison.net/2024/Dec/19/one-shot-python-tools/
https://news.ycombinator.com/item?id=42676432#42677178