biemster/FindMy #51: Implement Google Find My

biemster, 2024-04-10:
Google just activated their own Find My network on Android this weekend. Details are still scarce, so let's collect all technical info needed to send the BLE advertisement and subsequently query this network here.
The BLE part can then be implemented here: https://github.com/biemster/st17h66_RF (and I will also finally put the Apple FindMy advertisement there then as well, so the chip can do both)

darthnithin, 2024-04-10:
For anybody working on this the specification seems to be here (https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn)

trep-kalkyl, 2024-04-11:
This is very interesting.

1. Are there any indications that Google will release an API or documentation for this, or will a similar investigation like Haystack be needed?
2. Is there documentation on how the hardware is supposed to function and can it in that case be usefull for us?

darthnithin, 2024-04-12:
not sure yet. From that page, device manufacturers at least have to do a google NDA to ship a find my device compatible device, but that's pretty normal right?

humpataa, 2024-04-14:
My question is if Google / Android phones are now reporting Apple's FindMy devices to their (or even Apple's) server as well.   If Google isn't releasing info on how to request the data from their server, it probably needs to be hacked like it was done with Apple before. If Android now sends location reports to Apple's server it would enlarge the FindMy network instantly without the need to make any changes for Apple and us.

darthnithin, 2024-04-20:
Yea I'm fairly certain that's not happening. The google Find my Device network is separate. There are some architectural changes too. 
I think the only crossover between the networks is individual devices detecting unwanted trackers. 
[image]

rena2019, 2024-06-21:
Maybe helpful: from Nordic Semiconductor (manufacturer of nRF52840 - Multiprotocol Bluetooth 5.4 SoC supporting Bluetooth Low Energy, Bluetooth mesh, NFC, Thread and Zigbee)
Bluetooth Fast Pair: Locator tag (doc) https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/samples/bluetooth/fast_pair/locator_tag/README.html
Bluetooth: Fast Pair locator tag (zephyr source code) https://github.com/NordicBuilder/sdk-nrf/tree/0e99b70bcb259379b2faa87f9e2abcaf5de1e7a0/samples/bluetooth/fast_pair/locator_tag 
Fast Pair Validator (exclusively for manufacturers of bluetooth devices) https://play.google.com/store/apps/details?id=com.google.location.nearby.apps.fastpair.validator

dylanmazurek, 2024-10-14:
> I added an esp32 micropython script that sends the FMDN advertisement as detailed in https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn#advertised-frames table 8. I have no clue yet how to compute the ephemeral id or how to retrieve the reports through the google api's yet, but it's a start!
> 
> nRF Connect reports this as an Eddystone, with the ephemeral id as the data field.

I just got some Chipolo tags, if you need me to scrape some data that might help, happy to do so.

biemster, 2024-10-14:
> I just got some Chipolo tags, if you need me to scrape some data that might help, happy to do so.

That would be great! Are you able to mitm proxy an android with the account your chipolos are registered on?
(and confirming they advertise as an eddystone with a 20 byte data field would already be of huge help)

X-Ryl669, 2024-11-14:
> I have no clue yet how to compute the ephemeral id or how to retrieve the reports through the google api's yet, but it's a start!

I don't know if they improved meanwhile, but it's described [here](https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn#eid-computation)

As I understand the specifications, the Ephemeral Key is just a random 32 bytes generated by the tag once. It doesn't have to be known by the smartphone or tag owner. It's then used as an AES key (ECB mode!) for encrypting a dumb message. This makes a pseudo random number (`s`) that's used in the later crypto operations.

I'm not sure I'm getting the logic here, it seems so stupid, I'm missing something. The dumb message (which is doubled to fit in 256 bits) is made of only 22 bits of actual changing data. There's time's second counter whose 10 low bits are cleared (so only 22 bits useful in theory, but since it counts from 0, those are guessable quite easily). Technically, the counter actually update ~4 times per hour. 

Then you AES ECB encrypt it and get `r'`. There's a small micmac to build a 160 bits's X coordinate of the number on the curve using the usual ECC computations.

As for the ephemeral identity key, they say [this](https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn#read-provisioning-state), meaning that it's possible to fetch it if the tracker in not provisioned. In that case, everything is deterministic. 

If you ever intend to make a fully featured device (one that does support the provisioning process), then it means implementing [this](https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn#set-eik) (which is, AFAIU, not very complex to compute, but probably more to implement the persistent state in flash, in your reversed engineered firmware)

leonboe1, 2024-11-14:
This could be interesting to you as well:

- https://github.com/leonboe1/GoogleFindMyTools (Encryption, Decryption, Advertisement, Protobuf Reimplementation)
- https://github.com/dylanmazurek/google-findmy/issues/1

X-Ryl669, 2024-11-14:
The former repository is really useful, I didn't know it was possible to write crypto code over finite field that simply with python, always did it with C++ (and a not that simple interface). I've hard time figuring out what the second repo is actually doing (except making an HTTP request to `https://android.googleapis.com/nova/nbe_list_devices`). 

I think [this line](https://github.com/leonboe1/GoogleFindMyTools/blob/5d2323941e41aa33d4484311fe772bb182b89abb/FMDNCrypto/eid_generator.py#L13) is wrong however. 
They say in the documentation that you should clear the 10 low bits of the second / timestamp counter. That's because they are using a fixed key for encrypting the `eid` and user fetching the report should be able to query the expected report, so she must guess the content of `r` and thus the timestamp. By zeroing the 10 low bits, you have to guess a timestamp by steps of 1024 second, which is much easier that trying to guess a timestamp by step of 1 second.

While requesting location report is useful per se, it can be done actually with an Android phone meanwhile. Being able to have a tag that's speaking the right protocol so that an unmodified Android phone fills the location report's database at Google would confirm the protocol implementation in the tag is working. That would probably solve many problems already.

X-Ryl669, 2024-11-15:
Ok, I don't know in fact. [here](https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn#eid-computation) they say: `12 - 15 | TS[0]...TS[3] | Beacon time counter, in 32-bit big-endian format. The K lowest bits are cleared.`
The part I was referring to is **The K lowest bits are cleared**.

Seems like, to me that the beacon time counter should be `TS & ~1023`, since K is 10. So it should only have 22 bits of actual data, and incrementing every 1024 seconds. 
They say:
`Note: The device is assumed to have a 32-bit time counter in seconds.`
`Note: Rotation period exponent is fixed and set to 10, corresponding to 1024 seconds.`

Once you know it, you can recreate this EID just by knowing the time distance with the last or first provisioning (since everything is known in the algorithm). So you can compute Rx from both side (tracker & consumer's smartphone) which is required for encrypting and decrypting the EID (since you can deduce Ry from Rx). 

So if you don't mask the low bits, you'll not find the expected Rx from the consumer's smartphone that'll mask them out. (It'll only be correct once every 1024 seconds, but since the clock are unlikely to be synchronized, I don't think the tracker will emit a report that'll be captured by a nearby phone at the exact moment when its time counter is 0 for the 10 low bits).

leonboe1, 2024-11-15:
It's true that the counter for the EID calculation only counts in multiples of 1024. But recreating EIDs is only possible with the knowledge of the EIK.

X-Ryl669, 2024-11-15:
Well, I think we talk about different acronyms (in their immense wisdom, they used E for both ephemeral and encrypted).
The Ephemeral ID computation, which I've found in your code [here](https://github.com/leonboe1/GoogleFindMyTools/blob/5d2323941e41aa33d4484311fe772bb182b89abb/FMDNCrypto/eid_generator.py#L11) maps to [this](https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn#eid-computation), right ? (It looks like it does)

Yet, it's missing the part that's masking the low 10 bits, as specified.

It's not the encrypted ID yet, but it's required to compute the encrypted ID

leonboe1, 2024-11-15:
I've updated the file to make it more clear that only multiples of 1024 can be used. However, the file does specify the complete EID calculation. If you have more questions, please open an issue in my repository.

KieronQuinn, 2024-11-29:
I've done some digging into how the API works, perhaps it could be useful to you all.

The reason that there's no data in the response to requests to the `nbe_execute_action` endpoint is that they come as push notifications. The push data contains a payload (`com.google.android.apps.adm.FCM_PAYLOAD`), which is the actual response. This is as far as I got on this route, but the payload is in Protobuf format and seems to contain the encrypted location for a refresh action.

However, this may make using the API difficult. Libraries like [push-receiver](https://github.com/javajuice1337/push-receiver-v2) do allow for subscribing to push notifications for a FCM project, so might be useful here. A quick test using that library shows that using the app ID, project ID and API key from the Find Device app results in a push token, but this is only half the battle - the token still has to be sent to the server, which is a gRPC call that would need to be reverse engineered too.

As for getting an auth token for the server in the first place, you'll need an AAS token and to make a request using the right scopes using this token. The token can only be obtained via the embedded setup, there's various implementations of this (such as [Aurora Store's authenticator](https://gitlab.com/AuroraOSS/authenticator), [Revanced's MicroG-RE](https://github.com/WSTxda/MicroG-RE), or my own [Google Wallet plugin](https://github.com/KieronQuinn/SmartspacerPlugins/blob/main/googlewallet/src/main/java/com/kieronquinn/app/smartspacer/plugin/googlewallet/repositories/GoogleApiRepository.kt))

My original intention for working with the API was to do some basic location history logging to compare it to Samsung's network. In the end, I just made an [Xposed module](https://github.com/KieronQuinn/FindMyLog) that intercepts the data after the app has handled it. 

I've also done a lot of work in reversing Samsung's API, which is much easier to work with, that I'll write documentation for (useful for #65) when I release the project it's for - an app/mod that allows using Samsung's tags on non-Samsung devices.

leonboe1, 2024-11-29:
Hi Kieron,

Thanks for your report! I've also looked into querying the location reports for the last week. I'll have to agree that working with Google's implementation is indeed very annoying. Nevertheless, I'm happy to report that I made a breakthrough yesterday.

I am now able to log in with Firebase, get the required tokens, and receive the push notifications for the encrypted location reports in my own app. Root is not needed.

To make everything work, I used microG as a base, and added rudimentary support for the Spot API to it. My plan is to now to extract only the needed parts from microG. The project is very large, it can take up to 40mins to compile 😅

If I have compiled a more simple version, I'll post it on my GitHub. I'll keep you updated.

leonboe1, 2024-12-22:
Just in case you haven’t noticed already: The whole location querying process is now implemented in my repository (https://github.com/leonboe1/GoogleFindMyTools)

trep-kalkyl, 2024-12-25:
This is fantastic and a breakthrough. Could you please share some more information of this achievement?

> Just in case you haven’t noticed already: The whole location querying process is now implemented in my repository https://github.com/leonboe1/GoogleFindMyTools
> 
> I’ve also implemented an iOS app which does the same thing. If anyone is interested in testing this app, write me a message.

leonboe1, 2025-01-02:
Hi there, we will probably publish a paper with more details in the next months.

biemster, 2025-01-02:
> Hi there, we will probably publish a paper with more details in the next months.

That is great! As a fellow academic I can proofread if you need one (I usually have great difficulty finding proofreaders)

maelp, 2025-01-12:
Interesting, does this mean we can also create multiple "fake Google tracking tags" just like Apple FindMy, or would we still have to go through the regular Google "find my device" application process to create a trackable device?

maelp, 2025-01-13:
@leonboe1 Interesting, so I could do the same hack as for AirTag and generating fake keys, and then adding them to my ESP32 or whatever, and tracking them on the Google network? Can you point me at existing doc / code / repo for this?

biemster, 2025-01-13:
As far as I understood it can for now only retrieve reports for already registered real trackers right? And the fake tracker has to advertise a known-and-registered key? Please correct me if I'm wrong @leonboe1

maelp, 2025-01-13:
Interesting, can you give us further information on how to do this?

Message ID: ***@***.***>

biemster, 2025-01-13:
I did not find the time to test this out yet, I just try to follow the developments based on the commits as they come in :)

seoMattH, 2025-01-23:
@leonboe1 This will be really cool I'm pretty jacked about this integration

leonboe1, 2025-02-02:
Maybe next week - some things are annoying compared to Apple's network, but I might have a solution. I'll keep you updated.

leonboe1, 2025-02-08:
Hi there, I added a firmware for ESP32 trackers and code that registers them in your account in my repository. See the README file for details.

leonboe1, 2025-02-08:
Yes, you can only query location reports for trackers that were registered in your account. Google's network is quite different compared to Apple's.

leonboe1, 2025-02-09:
If you have an issue, create in an issue in the repository.

vasimv, 2025-02-09:
> Hi there, I added a firmware for ESP32 trackers and code that registers them in your account in my repository. See the README file for details. The firmware is very simple - let me know if it works for you.

Great work! I was able to adapt the code for nrf52 zephyr, but now i'm wondering - how to automate this, so a cron script would request data and put it in a database. It is possible to move authorization data to another machine (which doesn't have chrome)?

maelp, 2025-02-09:
Is the Google FindMy network hack working now? is there a tutorial somewhere on how to set up?

biemster, 2025-02-09:
> I still don't get why chrome needed, firebase messaging has its own API over HTTP.

I've looked into this a bit more over the last weeks, it seems it is quite tricky to get the necessary oauth token. every solution I could find either used selenium + chrome(ium or whatever floats your boat), or tried using curl / request or similar but was outdated and failed.

biemster, 2025-02-09:
> Is the Google FindMy network hack working now? is there a tutorial somewhere on how to set up?

I have yet to fully try this, but it definitely seems so. Please refer to https://github.com/leonboe1/GoogleFindMyTools for now as all the hard work is done by @leonboe1

leonboe1, 2025-02-11:
> > It is possible to move authorization data to another machine (which doesn't have chrome)?
> 
> This is exactly what I was wondering too, it would be an awesome feature! [@leonboe1](https://github.com/leonboe1)?

Yes, should be theoretically be possible by copying the secrets.json in the Auth folder. Why are you interested in this?

agittins, 2025-02-11:
> Yes, should be theoretically be possible by copying the secrets.json in the Auth folder. Why are you interested in this?

For me, my primary use-case is to have the data presented and tracked in Homeasistant. For my openhaystack tags I'm using https://github.com/malmeloo/hass-FindMy so I'd be looking for something similar for GFMDN devices at some point.

jcardus, 2025-11-04:
would be nice to have an hybrid firmware that advertised apple and google

jcardus, 2025-11-05:
Thank you @biemster!