biemster/FindMy #53: strict use of status byte since 17.5.1?

humpataa, 2024-06-02:
The documents show the first 6 bits of byte 2 of the advertising string being reserved, only the last 2 can be used "freely" (e.g. for battery state):

[image]

I've been using byte 2 of the advertisement fully for my own purpose which never happened to be a problem.
My phone has updated to 17.5.1 last night and today I see differences for faketags close to it.
Exactly since updating iOS the status byte keeps being set to zero. 

Interestingly the confidence byte also has dramatically changed exactly at the time of update: it shows values 4-5 times higher than before. Without tags or phone being moved a bit. Has anyone else noticed this?

humpataa, 2024-08-08:
I believe that beta updates of iOS (currently running 17.6.1) have fixed the problem. But I am using the status byte now according to the specification as shown above. Looking good.

humpataa, 2024-08-29:
not yet, unfortunately

bobrippling, 2024-09-12:
> The documents show the first 6 bits of byte 2 of the advertising string being reserved, only the last 2 can be used "freely" (e.g. for battery state):

@humpataa do you have a link to that document?