biemster/FindMy #58: ValueError: Invalid EC key

coolcodecding, 2024-08-21:
i have generated keys with the provided python script.
then i used the `./flash_esp32.sh ADVERTISEMENT_KEY`
after a few minutes i run request_reports.py but get an error

`200: 4 reports received.
Traceback (most recent call last):
  File "/home/myuser/work/openhaystack/findmy/FindMy/request_reports.py", line 88, in <module>
    eph_key = ec.EllipticCurvePublicKey.from_encoded_point(ec.SECP224R1(), data[5:62])
  File "/home/myuser/work/openhaystack/findmy/FindMy/.venv/lib/python3.10/site-packages/cryptography/hazmat/primitives/asymmetric/ec.py", line 183, in from_encoded_point
    return rust_openssl.ec.from_public_bytes(curve, data)
ValueError: Invalid EC key.`

Do you know what is the cause of this error and how could it be fixed?

coolcodecding, 2024-08-21:
after digging around i found this to work, although i dont now exactly why..
```
adj = len(data) - 88
eph_key = ec.EllipticCurvePublicKey.from_encoded_point(ec.SECP224R1(), data[5+adj:62+adj])
shared_key = ec.derive_private_key(priv, ec.SECP224R1(), default_backend()).exchange(ec.ECDH(), eph_key)
symmetric_key = sha256(shared_key + b'\x00\x00\x00\x01' + data[5+adj:62+adj])
decryption_key = symmetric_key[:16]
iv = symmetric_key[16:]
enc_data = data[62+adj:72+adj]
tag = data[72+adj:]
```

biemster, 2024-09-30:
duplicate of #52

closed 2024-09-30 by biemster