biemster/FindMy #61: Add support for TLSR825X

biemster, 2024-09-21:
Lidl Parkside batteries come with a tuya BT7L module, which have a nice flasher by @pvvx and an open source SDK + toolchain:
- https://github.com/pvvx/TlsrComSwireWriter
- https://github.com/Ai-Thinker-Open/Telink_825X_SDK

The examples/8258_feature_test is already configured for a simple advertisement only, and tested to work. Single step left is to implement the FindMy advertisement.

closed 2024-09-23 by biemster

Systm21, 2024-11-12:
Maybe it helps... 

https://x.com/atc1441/status/1847308917988085843

omarkhali, 2024-11-30:
> Maybe it helps...
> 
> https://x.com/atc1441/status/1847308917988085843

How to get public key

biemster, 2024-11-30:
> How to get public key

Please read the instructions carefully, it's all there. Also please don't intrude on multiple other issues, create a new one if you need help.

commit 06d3af26 (2024-09-23): Add tlsr825x firmware (closes #61)