biemster/FindMy #62: KeyError: 'M2'

lococafe, 2024-09-22:
Getting this new error with a recent rebuild of FindMy. Looks like Apples response has changed?

user@server:~/FindMy$ ./request_reports.py 
Apple ID: myapplieidhere@icloud.com
Password: 
pyprovision is not installed, querying http://localhost:6969 for an anisette server
pyprovision is not installed, querying http://localhost:6969 for an anisette server
Traceback (most recent call last):
  File "/home/user/FindMy/./request_reports.py", line 72, in <module>
    auth=getAuth(regenerate=args.regen, second_factor='trusted_device' if args.trusteddevice else 'sms'),
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/FindMy/./request_reports.py", line 33, in getAuth
    mobileme = icloud_login_mobileme(second_factor=second_factor)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/FindMy/pypush_gsa_icloud.py", line 37, in icloud_login_mobileme
    g = gsa_authenticate(username, password, second_factor)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/FindMy/pypush_gsa_icloud.py", line 90, in gsa_authenticate
    usr.verify_session(r["M2"])
                       ~^^^^^^
KeyError: 'M2'

biemster, 2024-09-23:
KeyError M2 means your login failed, are you sure you typed your apple id, password and 2FA correctly?

closed 2024-09-30 by biemster

viper-666, 2024-11-23:
Yes Apple-ID and PW are correct, else it hasn't add the Macbook to my account.
But there isn't a auth.json file.
[image]

biemster, 2024-11-23:
> Yes Apple-ID and PW are correct, else it hasn't add the Macbook to my account. 

Yeah indeed. Really odd, I'll have to look into this in more detail. Do you maybe have another account you could try this on?

biemster, 2024-11-23:
wut this went on another issue?

viper-666, 2024-11-23:
No, I have only this Apple account.

viper-666, 2024-11-23:
Oh, I found on a heise.de article that it could maybe seems that my Apple account hasn't enough Account-Score.
As I don't own a Apple device anymore and this account is old (from 2017) this may be the problem.
They say the best way is to use the AppleID on a newer iPhone.
I will look if I could find someone where I can use my AppleID on his phone.

biemster, 2024-11-23:
That is definitely possible, you could also just make a new id on account.apple.com

viper-666, 2024-11-23:
But than the Account-Score will never reach without using this account with a Apple device.

biemster, 2024-11-23:
i've had good experience with new accounts, they should have icloud properly enabled. I did however have issues with old accounts from before icloud times / early versions of it..

viper-666, 2024-11-23:
Ok, I have icloud on the old Account, I can log me in icloud with a browser.
And I can't make a new account, maybe because my phonenumber is already register in the old account.

viper-666, 2024-11-23:
I changed my password in Apple so the 2FA was asked again.
Here ist the output:
`{'dsid': '16822494xxx', 'delegates': {'com.apple.mobileme': {'status': 1, 'status-message': 'A server problem is blocking Apple ID sign in. Try signing in later.'}}, 'status': 0}`

biemster, 2024-11-23:
> I changed my password in Apple so the 2FA was asked again. Here ist the output: `{'dsid': '16822494xxx', 'delegates': {'com.apple.mobileme': {'status': 1, 'status-message': 'A server problem is blocking Apple ID sign in. Try signing in later.'}}, 'status': 0}`

That usually means something is off with your account, rarely a server problem (although it might be that ofc). Please try to get a new account, this is likely not an issue in the code.

viper-666, 2024-11-23:
I get a old iPhone and put my AppleID in, now all works like expected.
Thank you very much for your support and good work!