biemster/FindMy #67: Error when running "request_reports.py" for the first time

KlausMu, 2024-11-08:
I'm following the README. In step 3 I get:
```
./request_reports.py
Apple ID: ...
Password:
pyprovision is not installed, querying http://localhost:6969 for an anisette server
This implementation only supports s2k. Server returned s2k_fo
Traceback (most recent call last):
  File "/home/myuser/projects/airtag/FindMy/./request_reports.py", line 72, in <module>
    auth=getAuth(regenerate=args.regen, second_factor='trusted_device' if args.trusteddevice else 'sms'),
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/myuser/projects/airtag/FindMy/./request_reports.py", line 33, in getAuth
    mobileme = icloud_login_mobileme(second_factor=second_factor)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/myuser/projects/airtag/FindMy/pypush_gsa_icloud.py", line 39, in icloud_login_mobileme
    pet = g["t"]["com.apple.gs.idms.pet"]["token"]
          ~^^^^^
TypeError: 'NoneType' object is not subscriptable
```
Don't know what `This implementation only supports s2k. Server returned s2k_fo` means, but maybe this is the reason for the fail.

biemster, 2024-11-08:
This is new :( I'm going to have to look in to it..

biemster, 2024-11-11:
This problem is reported more widely, and there already seem to be a solution: https://github.com/malmeloo/FindMy.py/issues/62

KlausMu, 2024-11-11:
Oh, thanks, great. I applied the changes to file `pypush_gsa_icloud.py`, and indeed it works.

ZuriMuri, 2024-11-15:
Had the same issue and can also confirm that applying the mentioned changes to `pypush_gsa_icloud.py` in https://github.com/malmeloo/FindMy.py/issues/62 fixed it for me

biemster, 2024-11-19:
Fixed in 409888c

closed 2024-11-19 by biemster