biemster/FindMy #69: No 2FA code after entering the Apple ID

Kalle-79, 2024-11-18:
I made the adjustments in “pyphush_gsa_icloud.py.” After that, there are no more error messages. I entered the Apple ID and password. Now I am supposed to enter the 2FA code. Unfortunately, I am not receiving an SMS, nor is a code being displayed on any of my Apple devices.

When I log in online via a browser to Apple, both options work fine. I either receive the code on my devices or optionally via SMS.

Could it be that Apple has made some changes to the login system again? Has anyone else experienced this problem before?

Systm21, 2024-11-18:
There are so many issued that have been raised. There is no secret recipe, a user once made an PR where you could somehow select the device for the sms, try that.  I couldnt find it at this point.

Kalle-79, 2024-11-19:
Unfortunately I couldn't find it either. Maybe it has something to do with the fact that it's not a device that can be approved. I'll try again later and first delete all devices except my phone number. It's not nice to remove all the devices, but maybe it's possible.

biemster, 2024-11-19:
@Kalle-79 which changes to pypush_gsa_icloud did you make? If you have a mac you could consider running the scripts in the `monterey` branch on it, they just lift the search-party-token from the logged in user there.

Kalle-79, 2024-11-19:
I made the adjustments specified here in “pyphush_gsa_icloud.py” because the script not only gave the error the first time it was run, but always.

https://github.com/malmeloo/FindMy.py/pull/63/commits/261979b196f1d24a1bea26400bc92937a8eead49

After making the change it worked immediately and I was able to enter my Apple ID.
Unfortunately I only have an iPad and an iPhone available. The script runs for me under Ubuntu 24.04
I can't get a Mac up and running any time soon, but I'll see what I can do.

biemster, 2024-11-19:
What happens if you run `request_reports.py` with `--trusteddevice`?

Kalle-79, 2024-11-19:
Perfect, that was the solution. I now immediately received the code on both of my devices. Whatever the reason, it doesn't work via SMS.
Thank you very much for your help and for your work on this project! Now I can continue working on the project.

closed 2024-11-19 by biemster