biemster/FindMy #70: KeyError: 'service-data'` not a autification problem

viper-666, 2024-11-23:
In a other Issue you told that this error is a problem with the autification.
But autificatiton also the 2FA are accepted:

`Apple ID: XXX
Password: 
pyprovision is not installed, querying http://localhost:6969 for an anisette server
pyprovision is not installed, querying http://localhost:6969 for an anisette server
2FA required, requesting code
pyprovision is not installed, querying http://localhost:6969 for an anisette server
Enter 2FA code: 285869
2FA successful
pyprovision is not installed, querying http://localhost:6969 for an anisette server
pyprovision is not installed, querying http://localhost:6969 for an anisette server
pyprovision is not installed, querying http://localhost:6969 for an anisette server
Traceback (most recent call last):
  File "/root/FindMy/./request_reports.py", line 72, in <module>
    auth=getAuth(regenerate=args.regen, second_factor='trusted_device' if args.trusteddevice else 'sms'),
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/FindMy/./request_reports.py", line 34, in getAuth
    j = {'dsid': mobileme['dsid'], 'searchPartyToken': mobileme['delegates']['com.apple.mobileme']['service-data']['tokens']['searchPartyToken']}
                                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'service-data'`

Have you got a clue what is wrong?

biemster, 2024-11-23:
Could you print the whole `mobileme` response by adding a print call in the code there around line 34?

viper-666, 2024-11-23:
You mean like this, but there is a syntax error in it:

` print (mobileme = icloud_login_mobileme(second_factor=second_factor)
        j = {'dsid': mobileme['dsid'], 'searchPartyToken': mobileme['delegates']['com.apple.mobileme']['service-d>
        with open(CONFIG_PATH, "w") as f: json.dump(j, f) 
    return (j['dsid'], j['searchPartyToken']))`

biemster, 2024-11-23:
No sorry, I mean the output of `print(mobileme)` but then with the personal data removed from it by hand

viper-666, 2024-11-23:
I'm not a python expert, there is always a tab error???
[image]

biemster, 2024-11-23:
The file is space indented, please use spaces to indent as well instead of tabs.

biemster, 2024-11-23:
Also, please move the print statement a couple lines up, as it will error out now before it reaches it

viper-666, 2024-11-23:
It seems, the print don't work or have I to remove the device in Apple and run a fresh install of FindMy?
It doesn`t ask the 2FA again and in my Apple Account is a new Macbook:

[image]

`Traceback (most recent call last):
  File "/root/FindMy/./request_reports.py", line 74, in <module>
    auth=getAuth(regenerate=args.regen, second_factor='trusted_device' if args.trusteddevice else 'sms'),
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/FindMy/./request_reports.py", line 33, in getAuth
    mobileme = icloud_login_mobileme(second_factor=second_factor)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/FindMy/pypush_gsa_icloud.py", line 38, in icloud_login_mobileme
    g = gsa_authenticate(username, password, second_factor)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/FindMy/pypush_gsa_icloud.py", line 91, in gsa_authenticate
    usr.verify_session(r["M2"])
                       ~^^^^^^
KeyError: 'M2'`

[image]

biemster, 2024-11-23:
The `KeyError M2` is an authentication problem, it does not even reach your first print statement now. The fact that it does not ask for 2FA now is because apple accepted the MacBookPro as a valid device on your account, so that's good, don't remove it. A reinstall of FindMy is also not needed, it's just this script (although you could delete the auth.json file)

Are you absolutely sure your name:pwd combo is correct?

biemster, 2024-11-24:
continued and resolved in #62

closed 2024-11-24 by biemster