biemster/FindMy #72: Stuck with 2FA requests?

gmanic, 2024-11-26:
Maybe this is related to an old Apple Account with no trusted devices?

I am stuck with the 2FA - only getting this over and over again.
But, only with --trusteddevice I actually get a code to my phone (non Iphone) - nonetheless, after entering the code I just get the questions for the code again (but no new code being sent).

I'm trying to register a completely new account, this seems to have some hurdles as well (unsuccessful, so far)...

gmanic, 2024-11-26:
Will do
(try a new account and report back :) ).

gmanic, 2024-11-27:
This is really not what I would expect for as a procedure for prospective new customers....
But it's a pita....
Created account via android apple music, as it doesn't work on account-page at all, without any reason given. 
Go back to account-page. Login. Enter email-code. Enter phone-number. Enter sms-code, continue.
Next page offers only "continue" button. But it just doesn't work at all.
If I read the page source correct, there's a double </script> on line 258.

No means to tell anyone at that strange company....

Will keep trying. Sorry for the rant. :)

biemster, 2024-11-28:
I did not have any luck with music accounts either, and they are indeed not "full" accounts. However I've seen reports where people were able to convert them to proper accounts by adding payment options and such, but I never managed that either. pita indeed.

gmanic, 2024-11-30:
Time's quite scarce. Tried to add payment options to my (old) account, the (new) is still not working.
This is how modern world works, it seems. Not.
:-1:

gmanic, 2024-12-04:
Tried to log into icloud.com, which works.
"find my" does not work within firefox (forbidden by apple), but works on chromium.

There it says "no devices" and "to enable find my, log in with apple device..."

Still, no luck with 2FA request. I only get an sms if I use --trusteddevice, but afterwards I get another 2FA request and no sms.

gmanic, 2024-12-11:
Same here &  good idea :)

Got hold of an old iphone, 2FA successfully passed.

I'm closing this issue as everything of this great piece of work just works as expected (except apple).

closed 2024-12-11 by gmanic