biemster/FindMy #73: Public key

omarkhali, 2024-11-30:
Hi I have Telink TLSR-8252 How to follow the guide https://www.youtube.com/watch?v=RPGuPyxdHVw

It says Enter the FindMy/OpenHaystack Public key in the field

In this step, where do I find the Public key?

biemster, 2024-11-30:
This is @atc1441's awesome work, please ask for support there (although it does seem to rely on the firmware from here).
From a quick look at the video it looks like you need the hex of a public key, you could generate one using the `generate_keys.py` script here and convert the `Advertisement key` in the resulting `.keys` file to hex (so `b64decode(advkey).hex()`)

atc1441, 2024-11-30:
Thank you @biemster for clarifying it, indeed that is the key needed.
As mentioned in the video I did not want to go into that rabbit hole there and keep it simple^^
The public key field expects the "standard procedure"

biemster, 2024-11-30:
@atc1441 do you have the sources of your firmware available? If you used https://github.com/biemster/tlsr825x_FindMy you could mention that in the webflasher so people know where to look for support?
If you baked your own, I'd love to see the code ;) (since I'm not 100% sure of the power consumption of mine)

atc1441, 2024-11-30:
@biemster 
It is a custom firmware (Basically also just the simplest BLE example from Telink) with the added advertising code for the haystack which ends up at the ~9uA average.
The firmware is not public these days as it was just a side project for an evening^^ Don't have any setup running at all.

See attached a comparison of your and my version of the firmware so you can know more about the consumption.
Your TLSR does never sleep as it seems, after checking your code you seem to miss setting the "SuspendMask" so it is allowed to sleep.
You can add this line after the "blt_sdk_main_loop();" in the main loop:
`bls_pm_setSuspendMask (SUSPEND_ADV | DEEPSLEEP_RETENTION_ADV | SUSPEND_CONN | DEEPSLEEP_RETENTION_CONN);`
[image]
[image]

biemster, 2024-11-30:
Thanks so much @atc1441 for this comparison! I was wondering why my powertool battery (which was why I added this firmware) was draining quicker than a cr2032, now I know why! I did not want to pull out the bluetooth module from the battery pack, risking destroying the whole pack.
I'll add the code you suggest to my firmware when I find a bit of time next days, could I ask you to run a quick power consumption check when I finish that? (not urgent or anything, just when you are at your bench and accidentally have the setup in front of you)

Also @omarkhali I'm fully aware of the irony in asking you just an hour ago not to intrude on other people's issues with unrelated stuff, sorry...

omarkhali, 2024-11-30:
@biemster @atc1441 I am the one who should apologize for bothering you. I will try now in LYWSD03MMC install OpenHaystack firmware hoping it will work.

biemster, 2024-11-30:
No bother at all, I made this code public because I enjoy people trying it out. Good luck and let me know if it works!

omarkhali, 2024-12-01:
Hi @atc1441 @biemster 
I'm trying flash 
[image]

3.3V -> 3V
GND -> GND
TXD -> P14
DTR ->  RTS 

Using https://atc1441.github.io/ATC_TLSR_FindMy.html

Not working

biemster, 2024-12-01:
Can you try the flasher from this repo?

biemster, 2024-12-02:
Your connections are probably wrong, focus on that. It'll require a bit of experimentation, so don't give up easily. Also check if you provide enough power to the chip.

biemster, 2024-12-02:
@atc1441 could you do another run with your power meter on https://github.com/biemster/tlsr825x_FindMy/raw/refs/heads/main/out/FindMy.bin? I've added `bls_pm_setSuspendMask` in https://github.com/biemster/tlsr825x_FindMy/commit/ee5077af5eaef8760bf534f198ee89ff52b469fe, I hope that does the trick! (also I really should invest in a proper power measurement setup, are you using a nordic ppk2?)

biemster, 2024-12-02:
awesome thanks @atc1441 !