biemster/FindMy #74: Create our own tracker for $1!

biemster, 2024-12-07:
While it's fun to search the online mall for cheap bluetooth trackers, wait for a month for them to arrive and hope that they contain a flashable chip, let's instead make our own!

The target is very tight: **$1 in quantity 100+**

I've found a couple candidates already, but definitely would love to see some more suggestions in the comments.

Ideas so far:
1. InPlay NanoBeacon IN100 as suggested by @cyl0nius in #26
    https://inplay-tech.com/in100
2. AK1595 beacon tx only IC
    https://www.akm.com/us/en/products/bluetooth-low-energy-beacon/
3. Holtec Beacon transmitter BC7161 (likely needs a support mcu)
    by far the cheapest (~$0.2) if we can figure out how to omit the support mcu
    also might be difficult to source
    https://www.holtek.com/page/vg/BC7161
4. WCH CH592 / 582, CH32V208 or similar
    These are very cost effective (~0.5) and already have a findmy firmware here
    This would be my main focus
5. Telink tlsr825x
    Have very good support by @atc1441 and @pvvx for example
    but seem difficult to source

In conclusion I think the CH592 is the prime candidate due to it's availability at jlcpcb for example, and low price and passives count. However I would really like to experiment with the TX only AK1595 and BC7161 for example, since they would do only the bare minimum. Any ideas or suggestions are definitely very welcome, as are suggestions for a good fab house since this will be my first such project!

atc1441, 2024-12-07:
https://www.lcsc.com/product-detail/Microcontrollers-MCU-MPU-SOC_JieLi-Tech-AC6329C4_C5440081.html

biemster, 2024-12-07:
> https://www.lcsc.com/product-detail/Microcontrollers-MCU-MPU-SOC_JieLi-Tech-AC6329C4_C5440081.html

That one looks awesome, $0.35 even quantity 1! Have you ever programmed one of those?

atc1441, 2024-12-07:
Its flashable, but its has a very bad deep sleep :( So not good for a tracker

biemster, 2024-12-07:
aww that's a pity, I could not find that in the datasheet indeed

pvvx, 2024-12-07:
Cheap keys on ST17H66B are fully mastered (https://pvvx.github.io/iSearching/)
The [THB2](https://github.com/pvvx/THB2) project uses SDK for GCC, modified by me and [froloffw7](https://github.com/froloffw7), does not contain blobs. Has all sources in C for assembly. Full reverse engineering has been done.
The "KEY2" firmware contains an example of a ready-made BLE button for working in Home Assistant using the BTHome protocol. 
[image]

BLE OTA is supported in [PHY62x2BTHome.html](https://pvvx.github.io/THB2/web/PHY62x2BTHome.html).

atc1441, 2024-12-07:
Also while 1,50$ and not 1$ this is quite unbeatable in a DIY solution
https://youtu.be/RPGuPyxdHVw

biemster, 2024-12-07:
The ST17H66B is what started this project a couple years ago indeed :) Do you know any online pcb factories that still have them?

pvvx, 2024-12-07:
https://aliexpress.ru/item/1005004397030802.html
[image]
$1

pvvx, 2024-12-07:
The cheapest CH582M.
Broadcaster is the default test case from the WCH SDK. (Tested in 2022)
Broadcaster - BLE advertising only, no ability to connect or respond to scanning.
[image]
RF TX 0 dbm. In the example from the SDK changed: HAL_SLEEP=1, DEBUG - disabled, #define DEFAULT_ADVERTISING_INTERVAL 4800 (3 sec), DC-DC enabled.

On TSLR825x with similar parameters (3 sec, 0 dB) the average current is less than 5 µA, sleep is less than 1.8 µA (32 KB RAM is constantly active), the maximum peak during transmission is about 7 mA.

pvvx, 2024-12-07:
It is worth remembering that for beacons like iSearching and similar, the BLE advertising period should be 1 second. But there is not a single SoC that can last 1 year on CR2032 in this mode.
With pulse consumption (pulses up to 8 mA), the capacity of the average CR2032 is only up to 100 mAh. CR2032 batteries costing more than the key fob itself have a slightly higher capacity.

pvvx, 2024-12-08:
https://www.lcsc.com/product-detail/RF-Transceiver-ICs_PHYPLUS-PHY6222AAQC_C2836482.html

pvvx, 2024-12-08:
[TLSR827x Find My Network Accessories SDK Developer Handbook](https://doc.telink-semi.cn/doc/en/software/res/sdk/apple_findmy/tlsr827x_find_my_en/)

pvvx, 2024-12-09:
https://github.com/pvvx/THB2/issues/94

biemster, 2024-12-09:
That looks awesome @pvvx ! Very interested in the st17h66 BLE Long Range mode.

pvvx, 2024-12-09:
In the "Coded PHY" mode, the PHY chips have problems - the RF part slows down. The delays do not fit the specification. This affects the switching speed of the RF part TX-RX... But FindMy does not use reception.

biemster, 2024-12-09:
Do iDevices still register those broadcasts? I remember reading somewhere that even if an iPhone/iMac is able to do coded phy, it will not report findmy broadcasts sent in this mode.

biemster, 2024-12-09:
> And where can I find a full description of the format of the data transferred for FindMy? The available descriptions and the provided sources are more like a game of kindergarten.

I actually do not know of any official documentation of this, although oems can get "Works with Apple FindMy" for their tags so it must be somewhere?

pvvx, 2024-12-09:
I don't use Apple devices. None of them suit me in terms of functionality.
So I don't know anything and can't check. I don't even want to have Apple devices for testing.

atc1441, 2024-12-09:
> I don't use Apple devices. None of them suit me in terms of functionality. So I don't know anything and can't check. I don't even want to have Apple devices for testing.

Same :D 

The FindMy is only reverse engineered as there is no official doku available

biemster, 2024-12-09:
yup same here too :) although I did buy a mini for testing, which turned out to be useless

pvvx, 2024-12-09:
nRFConnect: 
[image]

Device data used:
[image]

The Key used is simply a random sequence of numbers.

pvvx, 2024-12-09:
A MAC address with 0xC0 (bits 7 and 6 in "1") in the first byte is a random MAC address. And it should be marked as Random in the BLE advertising data. But there is no mark in the BLE packet flags, if you look at the FindMy "childish writing".

biemster, 2024-12-09:
yeah they are really abusing the mac address to get some extra bytes there

pvvx, 2024-12-09:
But they don't put a mark in the flags that it's "Random MAC"?

biemster, 2024-12-10:
I was thinking to have something like this built: https://oshwlab.com/biemster/ch592tag

pvvx, 2024-12-10:
There is no button - without it, it is difficult to work with such a sensor. You can't update it, you can't change the key...
A ceramic antenna usually has a worse operating range than an antenna correctly routed on a printed circuit board.

biemster, 2024-12-10:
aw crap yeah I knew I forgot something!

biemster, 2024-12-10:
I made a new version with a PCB antenna, but I did not tune it or anything (because I don't really know how)

josemariaaraujo, 2024-12-11:
AFAIK unless you're a RF Wizzard you don't tune PCB antennas, you just copy proven designs.

atc1441, 2024-12-11:
Another idea.

This 1,6€ Remote features the ST16H77B
https://x.com/atc1441/status/1866798623654040014

biemster, 2024-12-11:
> AFAIK unless you're a RF Wizzard you don't tune PCB antennas, you just copy proven designs.

yes that's step one, but there is a matching circuit just before it that should be tuned to the feedline, usually called a PI circuit because of it's shape

biemster, 2024-12-11:
> Another idea.
> 
> This 1,6€ Remote features the ST16H77B https://x.com/atc1441/status/1866798623654040014

It's 4.50 for me? Do you have another source?

atc1441, 2024-12-11:
> It's 4.50 for me? Do you have another source?
It needs to be selected in the "3 for 3" offers but differs in price everyday^^
Today it is at 1,99€ for me.

[image]

biemster, 2024-12-11:
There are those bluetooth audio modules as well which are crazy cheap, like this: https://www.aliexpress.com/item/1005006151116762.html

Anyone here know of one that is programmable to some extend? or are they all OTP?

pvvx, 2024-12-11:
Cost of the case?
"Remote shutter"
[image]

pvvx, 2024-12-11:
Now I took it from the kids to see what's in there.
I don't know what kind of chip it is.
[image]
[image]

biemster, 2024-12-11:
anything on the uart pads?

pvvx, 2024-12-11:
[image]
[image]

olivluca, 2024-12-11:
I have a [similar one](https://github.com/olivluca/bluetooth-tiktok-remote/) (with more buttons) where they sanded the chip so it could not be identified but it turns out it's an otp part.

pvvx, 2024-12-11:
The cheap ST16H77B was sold as a LED strip control module (with the strip). Connecting USB charging.

biemster, 2024-12-11:
I've added the boot pin to a pad, should be quite complete now.

[image]

atc1441, 2024-12-11:
What about adding a real button and solder pads fitting for a simple CR2032 or CR2450 Battery holder ? :)

biemster, 2024-12-11:
A real button will take a lot of space, and is rarely used. I want to integrate that in the programming jig.
For power I don't know yet where the battery will go (under or on the side) so I left that out for now as well. GND pad is nicely on the side, I probably will add an extra pad for VCC there in the corner.

biemster, 2024-12-11:
Another idea I had is to go without a battery holder at all, and just epoxy the module and battery together. That way it's nicely waterproof too, and if I get the module to be cheaper than a battery they are almost disposable anyway :)

pvvx, 2024-12-11:
> That way it's nicely waterproof too, and if I get the module to be cheaper than a battery they are almost disposable anyway :)

In our stores CR2032 costs $0.08.

biemster, 2024-12-11:
> In our stores CR2032 costs $0.08. ("Fix price" set of 10 batteries $0.8.) 

I'm very jealous, I pay about 0.50 for them..

pvvx, 2024-12-11:
@biemster 

https://github.com/biemster/FindMy/tree/main/WCH_ch592

PPK2 (Shunt: switch): Sleep: 2.64uA 3.3V, Average: 5.42uA, Max Pulse: 19.15 mA (with PPK2 range switching surges)
Due to the automatic range switch in PPK2, dynamic load cannot be measured reliably.
[image]

My PowerProfiler (Shunt: stable 10 Ohm): Sleep: 2.6uA 3.3V, Average: 5.15uA, Max Pulse: 7.5 mA
[image]

Every 3.2 seconds, unused code is executed - unfinished firmware development.
[image]

The average CR2032 battery will last up to 1 year.
The 5 second beacon period is very long for many Bluetooth receivers. Exceeds all Apple recommendations.

pvvx, 2024-12-11:
@biemster

Not very fresh CR2032 battery and firmware https://github.com/biemster/findmy/tree/main/wch_ch592
Battery voltage:
[image]
Are you trying to repeat cheap Chinese fakes - do you propose to use the CR2032 battery with only 40%?

biemster, 2024-12-12:
> In many receivers, a single received beacon is not considered and is not recorded. A duplicate is needed, with a short period.
> 
> Linux with its quirks has a default maximum receive interval of 4 seconds :)

Thanks for the profiling! The firmware is not very polished yet indeed, I only added the advertisement to an example broadcaster.

I did not know that many receivers need a second advertisement to consider it valid, does that include one on another channel? Since the beacon is sent on all three advertisement channels?

pvvx, 2024-12-12:
2 advertising events. An advertising event is a broadcast on 3 channels.

For "connection" in Linux and Windows, receiving 2 advertising events is mandatory (!).
For Android - connection occurs upon first receipt of an advertising event.
Linux has the worst Bluetooth/BLE implementation and support of all operating systems.

Not everything is simple for receiving advertising. Some software has a fixed reception window. The reception window is often small - up to 4 seconds by default. If it is a multiple of the advertising event period and there is a processing pause, then there will be no reception. Such a program can also be in the adapter itself. But these are outdated adapter versions...

biemster, 2024-12-13:
I designed an even smaller one (only 12.45 x 12.45 mm):
[image]

I did not know pcb layouting is so much fun. I'll order these in a week or so, since I'm out for the holidays and the postman usually just sends stuff back then here.

pvvx, 2024-12-13:
There are already ready ones here: https://www.lcsc.com/product-detail/Bluetooth-Modules_Ai-Thinker-PB-03M_C2980078.html

biemster, 2024-12-13:
> There are already ready ones here: https://www.lcsc.com/product-detail/Bluetooth-Modules_Ai-Thinker-PB-03M_C2980078.html

Now you flippin tell me, you could have said this earlier? :D ah well it was fun, but I'm going to order these for sure! 0.60 in qty 10+ is something I can never top. Those phy6252 are supported by your firmware too right?

pvvx, 2024-12-13:
[PHY6252](https://github.com/pvvx/PHY62x2/blob/master/Doc/PHY6252_BLE_SoC_Datasheet_v1.3.pdf)

biemster, 2024-12-13:
I think we have the winner here.

pvvx, 2024-12-13:
Recently, PHY6252SD thermometers [TH05F_V1.4](https://github.com/pvvx/THB2/issues/96) have started to be sold on Aliexpress...

biemster, 2024-12-13:
> Recently, PHY6252SD thermometers [TH05F_V1.4](https://github.com/pvvx/THB2/issues/96) have started to be sold on Aliexpress...
> 
> And https://aliexpress.com/item/1005007656676765.html 100xPB-03M

Buying on LCSC is actually considerable cheaper than ali, that's a first for me.

biemster, 2024-12-16:
> Are you trying to repeat cheap Chinese fakes - do you propose to use the CR2032 battery with only 40%? [swra349.pdf](https://github.com/user-attachments/files/18102754/swra349.pdf)

If I replace the 2.2uF at VCC with two 47uF in parallel:
[image]
[image]

that would probably improve the situation @pvvx?

pvvx, 2024-12-16:
Capacitors are different. It is necessary to choose those with low leakage current. They are more expensive...

biemster, 2024-12-16:
> Capacitors are different. It is necessary to choose those with **low leakage current**. They are more expensive...

The murata X5R that LCSC found for me specifies an insulation resistance of 2000 MOhm*uF, which for a 47uF cap at 3 volts should be in the order of 0.1 uA right? Or am I missing some stuff?

[120 more replies]