biemster/FindMy #75: info and some questions

Philip2809, 2024-12-08:
First of, awesome project, when I found out about the openhaystack project i found it fascinating and wanted to play around with it. So thank you for this awesome project!

I am currently trying to build a nice ui to visualize the datapoints, I first tried to use an ESP32 to run the firmware and it worked great, I then got two low cost tags from Ali (https://www.aliexpress.com/item/1005003357986346.html) they had the ST17H66B and I was able to flash both and get them to report data, I have ordered some other that looks the same (but cheaper) from another seller, will see what shows up! Do you have any experience when buying from the same seller, is the chip usually the same?

I looked at your post about building a tag, this sounds interesting, I have thought of this, and an idea for it (from some tracker) it would maybe be nice to have a temperature sensor to know if the device is inside or outside.

Has there been any work done to make the button or buzzer work? I have not done anything in C so I am not so sharp at it. I also think the idea of making the buzzer sound when you send a ble signal to the device sounds cool, but how would this affect the power draw? I don't fully understand the sleep thing yet and I guess it could not be in a sleep mode if it is also listening for packets? I got the buzzer to make the faintest sound by using the blinky project but from my understanding you have to do some pwm stuff, not really sure at that yet.

biemster, 2024-12-09:
> First of, awesome project,

thanks!

> when I found out about the openhaystack project i found it fascinating and wanted to play around with it. So thank you for this awesome project!
>
> I am currently trying to build a nice ui to [...]

When you find a tag with a chip you like you should indeed by more quickly if you think you need them in the future. That gives you the highest chance of getting what you want, although sometimes sellers do change their product without notice, but it is still a higher chance then picking a new seller at random.

> 
> I looked at your post about building a tag, this sounds interesting, I have thought of this, and an idea for it (from some tracker) it would maybe be nice to have a temperature sensor to know if the device is inside or outside.

That's a great idea indeed! First iteration though will focus on cost, size and minimizing component count.

> Has there been any work done to make the button or buzzer work? I have not done anything in C so I am not so sharp at it. I also think the idea of making the buzzer sound when you send a ble signal [...]

I am not aware of anyone working on that, but the internet is full of examples in a non-tag situation. They should work the same. Having the tag listen for a ble signal will be detrimental for the power consumption though, since the chip indeed would not go to sleep (and even keep the whole RF stack warm). You'd need to replace the battery every few days if you stick with a cr2032.

Philip2809, 2024-12-09:
My idea is to order some more, but wanted to try a different seller as well because it was a lot cheaper, but I try one or two first before I order more. I havn't done any custom hardware and firmware coding but if I can help I would gladly do so. 

Yes, I will look into the buzzer but I understand about the power draw, it is not possible to only listen while doing advertising or that is too small of an timewindow to get any data? 

I have some questions about the st17h66, because I kinda like the idea of the battery service and want to see if I maybe can understand/implement something with it. Did you ever get gcc to compile the findmy or do you just use keil? Has there been any work on the key switching? I saw some posts about that this was worked upon but never found anything else?

biemster, 2024-12-11:
> My idea is to order some more, but wanted to try a different seller as well because it was a lot cheaper, but I try one or two first before I order more. I havn't done any custom hardware and [...]

That is indeed too small a time window, you'll miss the vast majority of messages if you want to save power by listening sporadically. Also you can't listen while sending the advertisement since the chip has only a single RF frontend + antenna

> I have some questions about the st17h66, because I kinda like the idea of the battery service and want to see if I maybe can understand/implement something with it. Did you ever get gcc to compile [...]

GCC works for findmy, I'll have a look if I forgot to push some changes or something.

> 
> edit: I tried to build with gcc, not working, tried keil but can't get it to work, i think because its too old. I have changed arm compailer to version 5 but still not working. Was able to get the code to work but then the linker did not work because of no FIRST/LAST, any ideas?

I used the old Keil, there are indeed issues getting the new one to compile the older SDK examples. About the battery service, since a couple days there is this: https://github.com/pvvx/THB2/issues/94