biemster/FindMy #76: Caching anisette generated header?

gmanic, 2024-12-26:
Today I ran into a 401 situation when calling request_reports.py. Usually it works just fine.
The error is not obvious, as the script just fails with a message of 'non-decodeable variable r.content.decode()'

Reason for the 401 was a restart of the VM with the anisette-v3 server on it. anisette-v3 always returns the same headers to all requests, as far as I can remember. But when restartet, the headers seem to change (could not verify on short notice), requiring running request_reports.py with '-r' to regenerate the access token (regenerating auth.json). This requires another 2FA (if you have no trusted-device), which in turn is somehow troublesome if you're away from the phone you're getting your code sent to.

Hence the question, if it would be possible to cache the generated anisette headers (indefinetely, or until an '-r' is issued) to avoid the regeneration and help unattended use? request_reports.py is running on a different VM on a different host anyway :)

biemster, 2024-12-26:
You most likely misremembered unfortunately, the anisette headers are different for every request. They can't be cached.

gmanic, 2024-12-26:
AH. My bad. Didn't look thorougly enough.

Closing as irrelevant.

Sorry for that.

closed 2024-12-26 by gmanic