biemster/FindMy #77: Sensor Date

sonman, 2024-12-30:
Hi, 

I want to include sensor data, especially temperature values, in the beacon. Where can I store this data, so that I can retrieve it later from Apple's iCloud?

I know there is a 2 bit battery status field. But that's too little.

biemster, 2025-01-01:
There is no space left in the advertisement for data. What you could do is use a different key for every temperature you want to measure, and just request reports for all these keys. Dividing your temperature range in 100 discrete values should be accurate enough, and querying for 100 keys is no problem either.
Since there is an unencrypted timestamp in the report, you could even encode your values in the public key since you don't need to decrypt the location!

sonman, 2025-01-01:
Thanks. 
An method you could encode your values in the public key I found here: https://positive.security/blog/send-my

closed 2025-01-07 by sonman