biemster/FindMy #78: How to compile the firmware for ESP32-S3

maelp, 2025-01-12:
I'm trying to run this on an ESP32-S3, is there anyway to compile the firmware for that architecture?

biemster, 2025-01-12:
The ESP32 compiled firmware is just copied from openhaystack so I never looked at that code, but the MicroPython script should run on the S3 although I did not test that since I don't have one. Could you try?

biemster, 2025-01-12:
wait I see now that it requires a custom build of micropython, forgot about that

maelp, 2025-01-12:
Okay I'll take a look!

BTW I'm trying to use the Apple FindMy micropython script instead for now, I copied the Advertisment key, but then I get

```
>>> from machine import base_mac_addr
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: can't import name base_mac_addr
```

(using the latest Micropython version)

biemster, 2025-01-12:
yes indeed, that `base_mac_addr` function is why you need a custom build. Making such a custom build is not complicated though, you could follow the instructions on micropython.org and then apply the patch that is already linked in this repo: https://github.com/micropython/micropython/files/10098686/modmachine_mac_addr.patch.txt

maelp, 2025-01-12:
Thanks! I'll take a look and report. So normally this code is functionnaly
equivalent to the OpenHaystack firmware, and I should be able to see my
device, right?

biemster, 2025-01-12:
yes exactly, this is basically openhaystack in python

maelp, 2025-01-12:
Okay, working now (it was a pain to compile micropython for some reason, plenty of bugs in the codebase not sure why)

closed 2025-01-12 by maelp

biemster, 2025-01-12:
great! could you share your fw for the s3?

maelp, 2025-01-12:
Here it is (this is a Micropython esp32-s3 distrib with the update for base_mac_addr)
[firmware.bin.zip](https://github.com/user-attachments/files/18391208/firmware.bin.zip)

olivluca, 2025-01-13:
Another possibility is to use esphome and the openhaystack component from [this repository](https://github.com/barrenechea/esphome-config-files).
(Or my [unmaintained fork](https://github.com/olivluca/esphome-config-files) that rotates through various keys).

maelp, 2025-01-13:
@olivluca does it work for esp32-s3 ?

olivluca, 2025-01-13:
Mmh, I'm not sure, I used it on an esp32-c3, see [here](https://github.com/seemoo-lab/openhaystack/issues/138#issuecomment-1207556369).

maelp, 2025-01-13:
Okay thanks!

I see in the firmware code that it uses a kind of "hardcoded" address for advertisment (that it calls "random address" yet it's hardcoded)

could that be an issue? should we just randomize it each time?

olivluca, 2025-01-13:
No, the "random_address" is actually part of the advertisement key and it is set by the set_addr_from_key function.

maelp, 2025-01-13:
Thanks!

Message ID: ***@***.***>

biemster, 2025-01-16:
Tags using this repo or any other of these simulated airtags need a custom app to retrieve the reports, they don't work with the official FindMy app.

maelp, 2025-03-15:
This was the shell script I used to flash it in case it could be useful

```sh
#!/usr/bin/env bash

cleanup() {
    echo "cleanup ..."
    rm "$KEYFILE"
}

# Directory of this script
SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null 2>&1 && pwd )"

# Defaults: Serial port to access the ESP32
PORT=/dev/ttyUSB0
PORT=/dev/tty.usbmodem1101

# Defaults: Fast baud rate
BAUDRATE=921600

# Parameter parsing
while [[ $# -gt 0 ]]; do
    KEY="$1"
    case "$KEY" in
        -p|--port)
            PORT="$2"
            shift
            shift
        ;;
        -s|--slow)
            BAUDRATE=115200
            shift
        ;;
        -h|--help)
            echo "flash_esp32.sh - Flash the OpenHaystack firmware onto an ESP32 module"
            echo ""
            echo "Call: flash_esp32.sh [-p <port>] [-s] PUBKEY"
            echo ""
            echo "Required Arguments:"
            echo "  PUBKEY"
            echo "      The base64-encoded advertisement key"
            echo ""
            echo "Optional Arguments:"
            echo "  -h, --help"
            echo "      Show this message and exit."
            echo "  -p, --port <port>"
	    echo "      Specify the serial interface to which the device is connected. (default: $PORT)"
            echo "  -s, --slow"
            echo "      Use 115200 instead of 921600 baud when flashing."
            echo "      Might be required for long/bad USB cables or slow USB-to-Serial converters."
            exit 1
        ;;
        *)
            if [[ -z "$PUBKEY" ]]; then
                PUBKEY="$1"
                shift
            else
                echo "Got unexpected parameter $1"
                exit 1
            fi
        ;;
    esac
done

# Sanity check: Port
if [[ ! -e "$PORT" ]]; then
    echo "$PORT does not exist, please specify a valid serial interface with the -p argument"
    exit 1
fi

# Prepare the key
KEYFILE="$SCRIPT_DIR/tmp.key"
if [[ -f "$KEYFILE" ]]; then
    echo "$KEYFILE already exists, stopping here not to override files..."
    exit 1
fi
echo "$PUBKEY" | python3 -m base64 -d - > "$KEYFILE"
if [[ $? != 0 ]]; then
    echo "Could not parse the advertisment key. Please provide valid base64 input"
    exit 1
fi

# Call esptool.py. Errors from here on are critical
set -e
trap cleanup INT TERM EXIT

# Clear NVM
esptool.py --after no_reset --port "$PORT" --chip esp32s3 erase_flash
esptool.py --chip esp32s3 --port "$PORT" write_flash -z 0 firmware.bin
```

and then I used the python file from the repo

```py
from machine import base_mac_addr
import binascii
import bluetooth

adv_key_b64 = 'BFY5W+2+fq30/gMzmS9vMoBAteIftzYq5miR2A=='
interval_ms = 300

public_key = list(binascii.a2b_base64(adv_key_b64))
if public_key[0] % 2:
    print('ERROR: The first byte of the public key has to be an even number,')
    print('due to limitations of the base MAC address on ESP32.')
    print('Setting the MAC address will fail.')
    print('To resolve this just generate a new key pair, since 50% of keys are affected')

findmy_apple = [
    0x1e, # Length (30)
    0xff, # Manufacturer Specific Data (type 0xff)
    0x4c, 0x00, # Company ID (Apple)
    0x12, 0x19, # Offline Finding type and length
    0x00, # State
] + public_key[-22:] + [
    0x00, # First two bits
    0x00, # Hint (0x00)
]

ble_mac = [public_key[0] | 0xc0] + public_key[1:6]
findmy_apple[29] = public_key[0] >> 6

base_mac_addr(bytearray(ble_mac[:-1] + [ble_mac[-1] -2]))

ble = bluetooth.BLE()
ble.active(True)
ble.gap_advertise(interval_ms *1000, adv_data=bytes(findmy_apple))
```