biemster/FindMy #80: Error: The script writes the wrong area of ​​Flash on ST17H66B!

pvvx, 2025-02-26:
The [script](https://github.com/biemster/FindMy/blob/main/Lenze_ST17H66/flash_st17h66.py) writes the **wrong area of ​​Flash** on ST17H66B!

[image]

Flash_st17h66.py log:
```
b'cmd>>:'
RESET MODE activated. Changing baudrate to 115200
Erase + Write
sent b'er512'
Response is: b'#OK>>:'
sent b'rdrev+'
Response is: b'0x01124485 6222M005 #OK>>:'
sent b'wrreg4000c890 ab000001 '
Response is: b'#OK>>:'
sent b'wrreg4000c838 ff010005 '
Response is: b'#OK>>:'
sent b'spifs 0 1 3 0 '
Response is: b'#OK>>:'
sent b'sfmod 2 2 '
Response is: b'#OK>>:'
sent b'cpnum ffffffff '
Response is: b'#OK>>:'
sent b'cpbin c0 002000 130 11002000'
Response is: b'by hex mode:'
sent c0 (len=304)
Response is: b'checksum is: 0x00010945'
sent checksum
Response is: b'#OK>>:'
sent b'cpbin c1 005000 3f50 11005000'
Response is: b'by hex mode:'
sent c1 (len=16208)
Response is: b'checksum is: 0x0015270a'
sent checksum
Response is: b'#OK>>:'
sent b'cpbin c2 020000 7088 11020000'
Response is: b'by hex mode:'
sent c2 (len=28808)
Response is: b'checksum is: 0x002a5483'
sent checksum
Response is: b'#OK>>:'
```
All ok.

But the recorded FW FindMy does not work!
After resetting the chip power supply, the previous FW starts, since FindMy recorded itself in the second half of the Flash!

And there are many such ST17H66B.
I'm not the only one getting this error.

biemster, 2025-02-26:
wut! how does it work for me? I'll look into this, thanks!

pvvx, 2025-02-26:
It's just a different revision of the chip.
With this chip revision your and my scripts do not work correctly.

biemster, 2025-02-26:
I gleaned those commands from the LeKit software, did you have a look how the latest version of this package is handling this chip revision?

pvvx, 2025-02-26:
I don't know what "LeKit" is.
There is a basic version of [PhyPlusKit](https://github.com/pvvx/PHY62x2/tree/master/PhyPlusKit).
All the same, they do not have a complete decoupling from the bootloader partition protection. And without this, you cannot make a copy - you cannot restore the original firmware, etc.

biemster, 2025-02-26:
> I don't know what "LeKit" is. There is a basic version of [PhyPlusKit](https://github.com/pvvx/PHY62x2/tree/master/PhyPlusKit). 

Yeah ofc 😄 that's what I meant.

So it's not a matter of flashing to the right flash area? I noticed `LePhyKit` generates some header for the provided binary with the right addresses to jump to, there is no playing with that?

pvvx, 2025-02-26:
It should be like this

[image]

(write BOOT_KEY2_v21.hex)

But it works until the power is turned off :)

biemster, 2025-02-26:
> This is either a ROM bug or a special chinese quirk. :)

Quite a quirk indeed :D I'd love to get one of those chips to try out, do we have a (semi) reliable source? If we have write access to the right flash area from our own code we might be able to cook up some minimal ota kind of contraption?

pvvx, 2025-02-26:
Such chips are installed in https://pvvx.github.io/SP621E/

I'm too lazy to make a flash loader in RAM for PHY... There is nothing complicated there. All ROMs of PHY chips have a command to write to memory...
But all commands in ROM are disabled when "FCT mode" is set (writing to a specific address of the Flash key to protect against copying). After that, only one command works - erase the entire Flash. Such devices have already been encountered... It is possible to write a new FW only by erasing all.

biemster, 2025-02-26:
i just received my first sp621e yesterday, i immediately ordered one after your discovery. I'll have a look if it has one of these quirky chips.

Do i understand correctly that after an "erase all" it still boots the original fw after a power cycle?

pvvx, 2025-02-26:
If you use ROM commands, then yes.
If you give the command to flash everything through the SPI controller registers, then it will erase everything :)
A running FW can easily write anything to the first Flash. It is protected only for commands in ROM (software).

pvvx, 2025-03-02:
The reason is described here
https://github.com/pvvx/THB2/issues/94#issuecomment-2688209817

pvvx, 2025-03-05:
1. After a power reset, the chip reads the eFuse and requires loading only encrypted firmware.
2. The programmer program writes zeros to the eFuse registers. After this, your firmware can be launched.

pvvx, 2025-03-05:
Updated version of "rdwr_phy62x2.py".
https://github.com/pvvx/PHY62x2/blob/master/Utils/rdwr_phy62x2.py
Using rdwr_phy62x2.py I did some experiments with the eFuse in the SP621E... 
I did FW with different attempts to rewrite eFuse. But nothing worked. eFuse are written only once, if they were not written before.

The resulting PB-03M modules have the 4th eFuse written, but this does not result in a state like the SP621E.

magnets110, 2025-03-30:
I have the MT1 remote with ST17H66B

```
=========================================================
PHY62x2/ST17H66B/TG7100B Utility version 01.03.25
---------------------------------------------------------
Connecting...
PHY62x2/TG7100B: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset Ok. Response: b'cmd>>:'
Revision: b'00124485 6222M005'
FlashID: 124485, size: 256 kbytes
PHY6222 - connected Ok
Flash Status: 0x0000
Flash Serial Number: 415034373733310b
Flash OTP: ffffffffffffffffff...
Flash 0x11002000: 06000000ffffffff3818ff1fffffffff
Flash 0x11002100: 00000111d00a000000000111ffffffff
eFuse Lock Status: 0x0000000f
eFuse 0: 0000000000000000
eFuse 1: 0000000000000000
eFuse 2: 0000000000000000
eFuse 3: a225c231a0676f99
Security Boot: 0
```

Will this be flashable?

pvvx, 2025-03-31:
You can try with any eFuse. If the original firmware is read, you can return it.