biemster/FindMy #82: request_reports no longer downloading data from multiple keys files.

swiss7, 2025-05-24:
As of yesterday, it seems that the request_reports script will no longer download data from all of the .keys file found in a folder. It seems that currently it only takes one of these keys files, with all of the others reporting as 'missing'. Yet the 'missing' one will download data if it is the only keys file in the folder.

Any ideas? Has something changed on the server side? Thanks!

swiss7, 2025-05-24:
It also seems to be limiting the downloads to 20 reports? Strange change of behaviour.

biemster, 2025-05-24:
Probably something changed on Apple's side, maybe only a single tag can be requested now? Report downloads were already limited to 20 when no startDate or endDate was provided (or something like that), it might be that that got changed as well. I'll have a look coming week, thanks for reporting!

vasimv, 2025-05-24:
My dirty hack: [onebyone.patch.txt](https://github.com/user-attachments/files/20426490/onebyone.patch.txt)

lovelyelfpop, 2025-05-25:
@vasimv If I request keys one by one, too many requests may cause http status code 401, then `Expecting value: line 1 column 1 (char 0)`

[image]

lovelyelfpop, 2025-05-25:
No wonder I can’t get the latest location for almost all the tags I made. This is because each request for multiple keys will only return 20 to 40 location reports, and these location reports may only belong to one of the keys.

lovelyelfpop, 2025-05-28:
I modified `request_reports.py`. I use async and parallel requests to save time. See:
https://gist.github.com/lovelyelfpop/9733899e3793df3b74d1da57095c46ce

olivluca, 2025-05-28:
Won't you get banned/throttled if you issue too many requests?

tobox, 2025-06-05:
Any news on this? Currently, I cannot get any results. Maybe apple has changed something on their side again?

The modified script of lovelyelfpop returns this:

> Request failed: BasicAuth() tuple is required instead
> Request failed: BasicAuth() tuple is required instead
> 0 reports used.
> found:   []
> missing: ['VMC4O6', 'Token1', 'Token3', 'Token2']
>

halex999, 2025-06-05:
The same result for me

sonman, 2025-07-03:
When sending 9 keys in one request, I get back reports for all of them. Total ~170 reports in response. Maybe if you have ~200 keys in a request only one will receive reports ?

aircable, 2025-07-03:
After some optimization runs, I found out, the longer the query the less number of reports the iCloud gives you. The best number was asking for 4 keys in one query, 4 queries at a time. Anybody else did those tests?

IDkonnecT, 2025-07-05:
@aircable Clear how you put several keys in one query, but how do you formulate several queries at a time inside same post request ?

biemster, 2025-07-05:
> It could be that Apple is throttling the keys for all users of the API, including official clients. I've just taken a look at some network logs of my Sonoma VM, and it's using a completely different API endpoint: `/findmyservice/v2/fetch` instead of `/acsnservice/fetch` as documented in the OpenHaystack paper. This API appears to only return a single location report per device though, which is quite the downgrade from what we have now.
> 
> Also, I'd like to urge everyone not to spam the API as a 'solution' to the problem; that might urge Apple to actively try to block 3rd party uses of the API and could ruin the fun for the rest of us.

Awesome, thanks Mike! I've been trying to find the time to revive my VMs for some captures but did not manage yet. Can you  give an example request json to the new endpoint, and did you implement this in your `FindMy.py` currently?

malmeloo, 2025-07-05:
I only managed to take a brief look at it, but I'll look into it again soon. From what I could tell it does appear to share more info about the actual device than before, including separation of the primary and secondary keys, and some device identifiers (?). I'm also not sure if they changed any authentication requirements or if it's just the same headers as before. Hopefully it won't be too difficult to migrate to the new API.

swiss7, 2025-08-15:
Is it possible to get a full working example (e.g. of request_reports.py) using the new API endpoint? I'm not quite sure what I'm getting wrong when trying to implement it. Thanks a lot!

biemster, 2025-08-15:
> Is it possible to get a full working example (e.g. of request_reports.py) using the new API endpoint? I'm not quite sure what I'm getting wrong when trying to implement it. Thanks a lot!

It's implemented in malmeloo's https://github.com/malmeloo/FindMy.py/pull/144, but it would be even more useful if you could share the error's you're getting? (I did not find the time yet to implement this, but your work will greatly help with that!)

biemster, 2025-08-19:
Can you print the `response.text`? (or maybe the whole response?)

swiss7, 2025-08-19:
Sorry for the hassle. The issue was with a single quote produced in the key (as opposed to double quote). I tried to install findmy.py but install failed (and documentation is rather minimal), but playing around with copying the code from L631 of account.py (from findmy.py) helped me a bit with solving the problem that I had.

Now it's returning a proper response, but still empty (0 records; using the original request_reports.py returns 20 records for exactly the same tag—the only tag in the folder):

>>> response
<Response [200]>
>>> response.text
'{"results":[],"statusCode":"200"}'

Anyways, I'll keep fiddling with it.

josephBoonrawd, 2025-08-25:
@jl-jung I got it

sonman, 2025-08-30:
Can your shortly describe your setup which you used to find out this ? Like VM, ... 
> Okay, I was not entirely correct yesterday. It does still appear to return multiple reports, my airtag was just too young.
> 
> The new API endpoint is `https://gateway.icloud.com/findmyservice/v2/fetch`. I can confirm that authentication works exactly the same way, so basic auth using dsid + searchpartytoken and anisette headers should do the trick.
> 
> Here's an example request body:
> 
> ```json5
> {
>    "clientContext":{
>       "policy":"foregroundClient",
>       "clientBundleIdentifier":"com.apple.findmy" // alternatively: "com.apple.icloud.searchpartyuseragent"
>    },
>    "fetch":[
>       {
>          "secondaryIds": [  // alt key name: "primaryIds"
>            // key IDs, same as before
>          ],
>          "keyType":1,
>          "endDate":1751846399999,
>          "ownedDeviceIds": [
>             // OF key IDs of account devices? just leave it empty
>          ],
>          "startDateSecondary":1751733308788,
>          "startDate":1751733308788
>       }
>    ]
> }
> ```
> 
> I observed two different `clientBundleIdentifier`s, it doesn't really appear to matter which one you use. They appear to separate key IDs into primary, secondary and owned device IDs now, but I'm not sure why. Fundamentally they're all the same anyway. Just don't put your custom tag's keys in the `ownedDeviceIds`, because those are listed separately in the response.
> 
> Example response:
> 
> ```json5
> {
>    "configVersion":1,
>    "acsnLocations":{
>       "locationPayload":[
>          {
>             "locationInfo": [
>                // encrypted payloads for the given key, format is stil exactly the same as far as i can tell
>             ],
>             "id": "" // hashed adv key
>          }
>       ],
>       "configVersion":1,
>       "statusCode":"200"
>    },
>    // this will be empty if you dont pass any ownerIds
>    "ownedDeviceLocations":{
>       "locationPayload":[
>          {
>             "locationInfo":[
>                {
>                   "locationTs":1751746897601,
>                   "location": "", // just an encrypted report again
>                   "fmt":0
>                }
>             ],
>             "id": "" // hashed adv key
>          }
>       ],
>       "configVersion":1,
>       "statusCode":"200"
>    }
> }
> ```
> 
> So there's not a whole lot that changed really, it's mostly just a new req/resp format. We are however missing the `description` and `publishedAt` fields now, but afaik the former has always been empty and the latter was not too useful anyway. The new API does appear to be a lot faster in my tests.
> 
> I also don't think they really fixed the start/end timestamp issue, in my tests they still just return all reports anyway. I haven't tested the limit on how many keys it will return yet.

jcardus, 2025-09-05:
> After some optimization runs, I found out, the longer the query the less number of reports the iCloud gives you. The best number was asking for 4 keys in one query, 4 queries at a time. Anybody else did those tests?

https://github.com/biemster/FindMy/pull/87

Chapoly1305, 2025-10-28:
I’ve noticed that queries that use either primaryIds or secondaryIds can return a maximum of 20 reports, and the response will always be a single key.

swiss7, 2026-04-14:
Hi all. Sorry to open this again, but has Apple just reduced the returns to 8 records per key?