biemster/FindMy #88: Apple's fetch api seems faster than before, but returns less location reports

lovelyelfpop, 2025-09-28:
This api(`https://gateway.icloud.com/acsnservice/fetch`) responses 60% faster than before, but the number of returned location reports has decreased significantly.

Half my tags' locations not get updated since one or two days ago.

And the new api(`https://gateway.icloud.com/findmyservice/v2/fetch`) returns empty response for half of my requests, even http status code is 200.

Did Apple change something again?

lovelyelfpop, 2025-09-28:
In the past, if a key was always with a certain Apple device, this Apple device would only report its location a few times in the beginning, and then almost never report it's location later. 

Now it seems that if a key has been used(has reported by any Apple device), it will not be reported by all Apple devices.

fz6, 2025-09-28:
> In the past, if a key was always with a certain Apple device, this Apple device would only report its location a few times in the beginning, and then almost never report it's location later.

I don't think is accurate? I have a friend with an iPhone who carries one of mine, his iPhone definitely seems to report the tag repeatedly. I've also got one in my back garden that would only be in range of a couple of neighbours, whose phones reliably report it in

biemster, 2025-09-28:
It's very likely that a software update for iOS is responsible for this, so it might be that the phones of your friend and neighbors did not update yet? I'm just speculating here, we'll have to see if we get more reports like this.

malmeloo, 2025-09-30:
I personally haven't experienced issues with a reduced number of location reports, but the empty response thing appears to be an issue on Apple's side. The FindMy app on macos is also running into it. The 'fix' is to just retry the request in a few seconds or so.

merasil, 2025-09-30:
I'm also experiencing this error in my application. The workaround suggested by malmeloo works, but very few reports are coming back. But if the FindMy app on macOS is also experiencing this problem, then it does indeed seem to be an Apple issue. I'm surprised that the problem has persisted for so long... isn't Apple usually quicker to fix these kinds of things?

biemster, 2025-09-30:
Apple is not quick on these things at all. When iOS 17 broke FindMy reporting completely (#40) , it took them to 17.4.1 to get it fixed. 😒